eez-open/studio #476: [LVGL] Adding literal buttonmatrix  functionality

elmonic, 2024-08-06:
Adding literal (add symbols) to matrix buttons increases their functionality.
[image]

liriyao, 2024-09-30:
A little bit question about buttonmatrix. we used to use valuechange event to get the active button index . it does't work nowadays.Is there any new way to do get the active btn index?

mvladic, 2024-09-30:
Yes. Check this issue: https://github.com/eez-open/studio/issues/510, in the comments there is:

> **Backward compatibly notice:**
>
> Before, events like KEY, GESTURE, ROTARY and VALUE_CHANGED (in case of ButtonMatrix widget) were sending some specific data through event handler connection line (for example GESTURE even was sending gesture direction). Now, this is changed so that these events also sends event object, so for example you need to use Event.getGestureDir(event) to get the direction for GESTURE event.

So, for ButtonMatrix to get btn pressed here is how event handler is used:

[image]

liriyao, 2024-09-30:
> for ButtonMatrix to get btn pressed here is how event handler is used:

OK,thanks for your work ,it helps a lot