eez-open/studio #479: Generated image c file must use project defined lvglInclude parameter

drik, 2024-08-08:
**Describe the bug**
When I define LVGL Include parameter in my project, EEZ still generate c file with lvgl/lvgl.h 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Project Settings --> Build --> LVGL Include (change tthe default 'lvgl/lvgl.h' by 'lvgl.h'
2. Add an image (named xxx) to your project and Build
3. Open the generated ui_image_xxx with a text editor
4. On line 7, you can see #include "lvgl/lvgl.h"

**Expected behavior**
The build system must use LVGL Include parameter to generate ui_images

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: 11

** LVGL version (if used)**

- 9.x

mvladic, 2024-08-08:
We are using [LVGL tool](https://github.com/lvgl/lv_img_conv) to generate these image files and doesn't change them afterwards. If you look closely you can see that following is generated:

```
#ifdef LV_LVGL_H_INCLUDE_SIMPLE
#include "lvgl.h"
#else
#include "lvgl/lvgl.h"
#endif
```

So, you can add preprocessor definition `LV_LVGL_H_INCLUDE_SIMPLE` and `lvgl.h` will be included.

petrubuciu, 2024-09-12:
i had to ask because i have the same problem, where to make this definition of LV_LVGL_H_INCLUDE_SIMPLE?
For now i'm editing every image_xx.c file to include the "lvgl.h" and not "lvgl/lvgl.h"
Sorry if my question is noob. thx in advance.

mvladic, 2024-09-12:
> where to make this definition of LV_LVGL_H_INCLUDE_SIMPLE

Are you on Arduino?

petrubuciu, 2024-09-12:
ESP32 s3

mvladic, 2024-09-12:
I think you can't add preprocessor definitions using Arduino IDE.

So, I decided to prepend each generated `img_....c` file with the following:

```
#ifdef __has_include
    #if __has_include("lvgl.h")
        #ifndef LV_LVGL_H_INCLUDE_SIMPLE
            #define LV_LVGL_H_INCLUDE_SIMPLE
        #endif
    #endif
#endif
```

This should fix the issue for Arduino.

mvladic, 2024-09-12:
Implemented as described in my previous comment.

closed 2024-09-28 by mvladic