eez-open/studio #482: [LVGL] Chart has no personal settings

elmonic, 2024-08-10:
Hi
It is better to add settings like below to the chart.
If its parameters can be adjusted so that we have a live display, it will be very good.
Change Setting:

[image]

Change X , Y axis Setting

[image]

add Data to chart

[image]

elmonic, 2024-08-11:
hi
Will it be added in the next version?

mvladic, 2024-08-11:
Yes, there is a good chance that this will be done in the next release.

littleboot, 2025-05-08:
I was/am also looking for this functionality, I switched from squareline studio and was trying out everything in EEZ Studio I thought the chart widget was broken until I found this issue report (Its just not fully implemented yet, all I wanted to know).