eez-open/studio #486: Micropython support

kdschlosser, 2024-08-16:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Full support for generating code that will run on an LVGL MicroPython Binding.

I am not talking about the official binding as it doesn't have a consistent API for displays and input devices.

I have been spending a pretty sizeable amount of time developing a new MicroPython binding to LVGL. I have made it easier to compile the firmware and added quite a few display drivers and touch drivers. Most importantly I have added support for I8080, RGB and LED Matrix's. I also reworked the SPI interface so SPI touch devices as well as SD Card can be sued on the same SPI bus as a display.

There is a static API across almost all display drivers. There are only a few that are only a tiny bit different and those are for less common displays. The touch drivers are pretty consistent across them, some might have additional features that can be used but these are optional. Because of the consistent API adding full support is easier to do because a single code output template would cover 95% of the displays and a single template would cover 100% of the touch drivers.

I can even go to the extent of generating a JSON file that contains the entire LVGL API for the binding complete with data types and documentation too. That can be used to dynamically create the different controls in your software.

The binding can also be compiled to run on macOS, Windows and Linux and the same code that runs on an MCU is able to be run on the desktop. So you can provide a preview window of the GUI actually running real time as changes are made.

**Describe alternatives you've considered**
The "other" studio (ack!)

**Additional context**
None

kdschlosser, 2024-08-16:
stupid me, I forgot to add a link to the repo...

https://github.com/lvgl-micropython/lvgl_micropython

mvladic, 2024-08-16:
This is really interesting. 

I assume that only the latest LVGL version is supported, i.e. LVGL 8.4 is not supported with this repository?

Have you tried to build it with Emscripten? Studio can only use web assembly to run LVGL code for the editor preview and run simulation.

One advantage over the C version is that the user could write what we call native UserAction's using MicroPython and that code could be executed in the Studio simulator, which is not possible with the C code since studio doesn't come with C compiler.

kdschlosser, 2024-08-16:
Currently LVGL 9.1 is supported. Because of how I wrote the binding it would be easy to add the ability to use different versions down to 8.3. If that is something that would be wanted.

I would need to create a second set of driver frameworks because the API is different between LVGL 8.x and 9.x but it wouldn't be that hard to do...

closed 2026-09-16 by kdschlosser