eez-open/studio #487: [LVGL] Is or will there be an option to group widgets?

O491dogan, 2024-08-16:
**Is your feature request related to a problem? Please describe.**
Not directly a problem, rather being lazy. I'm using an M5Dial and eez to create the gui. The M5Dial has an builtin encoder which I'd like to use with lvgl.
The problem is that I (at least I do think so) have to include the _screens.h_ in my main program and group the elements by myself.

**Describe the solution you'd like**
It would be nice to select the widgets we want to group and then right-click and hit "_group..._". The new group then shows up  in the left where pages and so on are lcoated in a new column called _"groups"_.
The so created groups are grouped in lvgl like this:
```
lv_group_t *group = lv_group_create();
lv_group_add_obj(group, btn1);
lv_group_add_obj(group, btn2);
lv_group_add_obj(group, sld1);
```

And can be added a input device (maybe also in the settings of the created group) like so:
`lv_indev_set_group(encoder_indev, group);`

**Describe alternatives you've considered**
Alternatively one can do it manually as described above.

**Additional context**
I'm pretty new with eez studio, like it very much and it just works... simply amazing. So please be kind if I have missed an option that is already related to it.
Thank you for this piece of open source software!

mvladic, 2024-08-16:
Same as #229

mvladic, 2024-09-23:
I'm closing this issue and continue to work on this feature under issue #229.