eez-open/studio #489: [LVGL] add action Increment and Decrement value Spinbox

elmonic, 2024-08-18:
The spinbox widget has the ability to increase and decrease the value with the selection of a number in action LVGL.
If possible, it should be added to the software.

void **lv_spinbox_increment**([lv_obj_t](https://docs.lvgl.io/master/API/misc/lv_types.html#_CPPv48lv_obj_t) *obj)
void **lv_spinbox_decrement**([lv_obj_t](https://docs.lvgl.io/master/API/misc/lv_types.html#_CPPv48lv_obj_t) *obj)

[image]

Similar to the image below:
[image]

mvladic, 2024-08-18:
Check the Spinbox example:

[image]

Here the spinbox value increment and decrement is implemented by simply changing the value variable.

elmonic, 2024-08-18:
thanks I saw the example that you posted, but this method is not practical and has limitations, that is when the number of widgets of the spinbox increases and this process must be done for each one.
This feature of increasing and decreasing, shifting and the selected digit is placed inside the LVGL itself and is compatible with the mouse and touch as well as with the keyboard keys of this feature (such as up, down, and enter).

[image]

I did this process by coding, but it is much better if it is done with the software itself and it increases the speed and efficiency of the design.
[image]

mvladic, 2024-08-18:
I don't understand the argument that when the number of Spinbox's increases the meted presented in Spinbox example is not practical.

elmonic, 2024-08-19:
When you are working on a large project and you have, for example, 20 spinbox widgets, for each of which, according to your example, we have to put 6 flow actions and we have to create a total of 120 actions inside the page, do you think it makes sense?! When LVGL itself supports this process, why do we start the path again and this process occupies more RAM.
All these conditions are new without taking into account the limitations of choosing and moving the character that I wrote to you before.

Look at the page below. Now, if I want to do your process for 10 spinbox widgets, do you think it is reasonable with this number of actions?
[image]

elmonic, 2024-08-19:
Like the video below, very easy and fast😊
https://github.com/user-attachments/assets/d74c899c-3988-45c1-992d-729e3cfa8e52

mvladic, 2024-08-19:
While tinkering about this issue I again tested the Spinbox example from EEZ Studio. In my example, when Spinbox step is changed with the touch interface then step variable in the flow is not properly updated (this is the case you showed in your video). But, if you change the step with the encoder everything works as it should. I reported this difference in behavior in LVGL repository, see https://github.com/lvgl/lvgl/issues/6702. 

Ok, I can add these two actions in the future as you recommended.

> When you are working on a large project and you have, for example, 20 spinbox widgets, for each of which, according to your example, we have to put 6 flow actions and we have to create a total of 120 actions inside the page, do you think it makes sense?!

Actually, only two flow actions are required:

mvladic, 2024-08-19:
Still, to be fair, my solution requires another variable for the step. If we have action for Spinbox Increment and Decrement then we wouldn't need this variable.

ttump, 2024-08-20:
I understand there is a bug in LVGL but should this work within EEZ Studio testing? I've tried the spinbox example but clicking in the spin box those not update the step variable. The highlighted number reverts to the right digit and step remains at a value of 1. Windows version 0.15.1

mvladic, 2024-08-20:
It is precisely because of this bug that it does not work in Studio.