eez-open/studio #492: [LVGL+Flow] Spinbox specific properties can be expressions/assignable

fietser28, 2024-08-23:
**Is your feature request related to a problem? Please describe.**
I can't dynamically change some of the properties of the spinbox.

**Describe the solution you'd like**
I would like to have the option to set the specific settings of the spinbox to an expression, not just literal.
These are the settings that the use of the spinbox doesn't change (expression not assignable).

In my case the spinbox properties need to change when I change to a different mode in my application. As an example I've filled in the expressions I need in the literal fields:
[image]

**Describe alternatives you've considered**
1. Create a fixed spinbox for each mode and swap between these.
2. Keep my current manual implementation in flow/native.

**Additional context**
This has a relation to #229 because spin boxes are mainly useful with encoders (imho).

fietser28, 2024-08-23:
I noticed LVGL API is lacking:
- lv_spinbox_get_range_min
- lv_spinbox_get_range_max
- lv_spinbox_get_digit_format_count
- lv_spinbox_get_digit_format_position

So did some digging in lvgl sources. Would this be an option to use (to get min range value) as a temporary(?) workaround:
```
        lv_spinbox_t *spinbox = (lv_spinbox_t *)objects.obj14;
        int32_t cur_value =  spinbox->range_min;
```

fietser28, 2024-08-23:
I've implemented part of it. Min/max range can be an expression now. 
See branch: <https://github.com/fietser28/studio/tree/spinbox>

Will work on the format count and position options in the same way before creating a merge request.

Feel free to comment on the solution I used.
I might also create an issue in LVGL repo whether they will add the above calls. Probably won't be added backwards to v8.

fietser28, 2024-08-25:
[image]
is working now on MCU using my branch. The simulator is WIP.

mvladic, 2024-08-26:
I flew through the changes a bit and it looks pretty good to me. When you're done, I'll take a closer look. The next release should be by this Monday. If you can finish it by Friday then we can add it to this release.

mvladic, 2024-08-28:
> I noticed LVGL API is lacking:
>
> * lv_spinbox_get_range_min
> * lv_spinbox_get_range_max
> * lv_spinbox_get_digit_format_count
> * lv_spinbox_get_digit_format_position
>
> So did some digging in lvgl sources. [...]

FYI In version 9.2 you can't access members of `lv_spinbox` or any other widget directly. Workaround is to include "lvgl/src/lvgl_private.h".

fietser28, 2024-08-28:
> FYI In version 9.2 you can't access members of `lv_spinbox` or any other widget directly. Workaround is to include "lvgl/src/lvgl_private.h".

Thanks for the tip. 

I don't know if I will have the time to finish this before the end of the week.

mvladic, 2024-08-28:
> I don't know if I will have the time to finish this before the end of the week.

Never mind, take your time and we will release it when it is ready.

fietser28, 2024-09-28:
I've added some updates to the branch (while following master branch). Current status is:

In simulator:
- V9.1 works, except when step is an assignable: Then clicking on a digit with mouse the position flips back. 
- V8.3 works until you click on the spinbox: Then simulation freezes. Can't figure out why.
- AFIAK generated code works also in combination with groups.

fietser28, 2024-09-28:
Here is a test project to explore the different features. 
[spinboxTest.eez-project.zip](https://github.com/user-attachments/files/17174550/spinboxTest.eez-project.zip)

fietser28, 2024-09-28:
Also note the last commit I make [Commented out unnecesairy callbacks: These are expressions not assignable?
](https://github.com/fietser28/studio/commit/2c8c8e83e552fb507a941490a4d43c18ad71b7cb). 
Is the assumption correct these callbacks are not needed/must not be present when the parameter is an expression?

mvladic, 2024-09-30:
I'm not happy how currently adding new LVGL widget works. We have separate definitions for editing and running inside simulator (`lvglCreate` functions) and for source code generator (`lvglBuild` functions). So, you need to implemented all these functions:

```
lvglCreateObj
lvglBuild
lvglBuildTickSpecific
```

and some more:

```
getIsAccessibleFromSourceCode
lvglPostBuild
```

And, even worse, is that for `lvglCreate` you need to implement wrapper code for the LVGL (the stuff in `studio_api.cpp`).

I think, ideally, you should implement only one function that should work for both simulator and source code generator. And you shouldn't need to write wrapper code.

I have some ideas how to do it, so for now I think you should stop working on this since it is possible that very soon I start to refactor all this `lvglCreate` and `lvglBuild` code.

fietser28, 2024-09-30:
Ok, I agree with you on your analysis. For adding/changing a single widgets parameters I had to touch code in a lot of different places (was not easy to figure it out).

I also changed a few things in [expression-property.tsx](https://github.com/fietser28/studio/blob/spinbox/packages/project-editor/lvgl/expression-property.tsx). Imho these functions will become unmaintainable if more LVGL versions and widgets are added.

fietser28, 2025-03-16:
After the 'big refactoring of the LVGL code generator and LVGL simulator' will it be relatively easy to add min/max/step/digit count support for spin boxes by only editing the Spinbox widget file?

mvladic, 2025-03-16:
Yes, it should be easier. Now, only one function `toLVGLCode` must be defined per widget and this function is used for both simulator and code generator.