eez-open/studio #495: [EEZ-Studio] Dim or hide lines to hidden widgets

fietser28, 2024-08-26:
**Is your feature request related to a problem? Please describe.**
In screens/dashboards where widgets are overlays I often hide the widgets (hidden in the widget structure panel, not the hide property of the widget itself) to unclutter the screen during development.
The lines to these hidden widget are still visible.
This issue is for convenience / nice UI only.

**Describe the solution you'd like**
Dim or hide the lines to widgets hidden in the widget structure. 
It would be nice if the user can choose between dim and hide for the lines.

**Describe alternatives you've considered**
Live with the clutter.

**Additional context**
Widget hidden, lines are floating. Dimming or hiding the these floating lines is the scope of this issue.
[image]
Widget shown. The lines are now connected to the shown widget handles:
[image]

mvladic, 2024-09-23:
Implemented.

Dim or hide option can be selected here:

[image]

or in Project Settings, where you can also change the opacity of the dimmed lines:

[image]

fietser28, 2024-09-23:
It is working fine. Thank you!

I do have a text suggestion: 
Add the text "Hidden widgets lines" left of  the rolldown in the widget structure (and the rolldown can be smaller):
[image]
If the text doesn't fit because the panel is to small the text can be clipped.

mvladic, 2024-09-23:
Changed as suggested.

closed 2024-09-26 by fietser28