eez-open/studio #501: Direction option for LVGL Dropdown widget

vldgeorgiev, 2024-08-28:
**Is your feature request related to a problem? Please describe.**
The Dropdown widget (lv_dropdown) supports a Direction property for placement of the list (up, down, left, right). The option is not supported by EEZ Studio.

**Describe the solution you'd like**
It will be useful if EEZ Studio ccan control that property

[image]

mvladic, 2024-08-30:
Implemented.

vldgeorgiev, 2024-09-02:
@mvladic Thanks for the quick implementation. I tried it and it does work, the list  appears on the side, but there is a glitch.
The list in my case is larger (higher) than the screen and the top part gets cropped. I can scroll it, but can't select or see the top 1-2 items in it. Is that an LVGL issue or something adjustable from EEZ Studio?
[image]

zalexzperez, 2024-09-02:
There's another  little problem.
Both the BOTTOM and TOP directions share the same indicating arrow pointing downwards.

closed 2024-09-03 by prasimix