eez-open/studio #516: [LVGL] Event callbacks uses incomplete `lv_event_t` typedef

technophil98, 2024-09-03:
**Describe the bug**
Since 0.16.0 (#457), `lv_event_t` user data is configurable from Studio. Generated code cannot compile:

```c
static void event_handler_cb_abc(lv_event_t *e) {
    lv_event_code_t event = lv_event_get_code(e);
    if (event == LV_EVENT_SCROLL_END) {
        e->user_data = (void *)42;
        action_abc(e);
    }
}
```

Compiler error:
```console
error: invalid use of incomplete typedef 'lv_event_t'
   17 |         e->user_data = (void *)42;
      |          ^
```

**To Reproduce**
1. Create a LVGL component
2. Add an event handler
3. Generate code
4. Compile should fail

**Expected behavior**
User data value should be passed to `lv_obj_add_event_cb` call:
```c
lv_obj_add_event_cb(obj, event_handler_cb_abc, LV_EVENT_ALL, (void *)42);
// 42 should be passed here instead of the callback function --^
```

**Workaround:**
Downgrade to EEZ Studio v0.15.0 or earlier

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version: 0.16.0
 - Compiler: GCC 12.4.0 (Cygwin)

**LVGL version:**
- 9.2

mvladic, 2024-09-04:
Fixed.

mvladic, 2024-09-04:
This fix is now released in [0.16.1](https://github.com/eez-open/studio/releases/tag/v0.16.1).

technophil98, 2024-09-04:
Wow that was fast 🚀
Thank you!

closed 2024-09-04 by technophil98