eez-open/studio #519: Code generation for images does not honour 'LVGL include' build setting

bentron2000, 2024-09-05:
**Describe the bug**
When generating code from imported png bitmaps, the generated image.c code imports via `#include "lvgl/lvgl.h"` even if the LVGL include path has been modified.

**To Reproduce**
Steps to reproduce the behavior:
1. Add an image to a screen
2. In Settings/Build - change the LVGL include path to something specific like `testing/lvgl.h`
3. build
4. Review generated C code for the image - the include path will not be the one set in the LVGL include ie. `testing/lvgl.h`

**Expected behavior**
The generated code should honour the parameter set in setting/build

**Screenshots**
n/a

**Desktop (please complete the following information):**
 - OS: Mac OS 14.4

** LVGL version (if used)**
9.1

mvladic, 2024-09-05:
Also reported in #479

mvladic, 2024-09-13:
I'm closing this because it's a duplicate.

closed 2024-09-13 by mvladic

zggcd157, 2024-12-29:
I also had this problem when using EEZ 0.2x, I modified it to LVGL.h, but the resulting inclide is still lvgl\ LVGL.h