eez-open/studio #531: [LVGL] EEROR after export (userData)

elmonic, 2024-09-11:
hi thanks for new version V16.1
I have eeror when export LVGL project 
[image]

mvladic, 2024-09-11:
We can only associate integer values (#457) as user data to event handler, therefore cast from `void *` to `int32_t`.

mvladic, 2024-09-11:
Can I see which error you are getting?

elmonic, 2024-09-11:
hi
The output of the lv_event_get_user_data function is of pointer type and the compiler cannot cast it to an integer number.
my eeror:
cast from 'void*' to 'int32_t' {aka 'int'} loses precision [-fpermissive]

LVGL V9.2
IDE Eclipse CDT V2024

elmonic, 2024-09-11:
Now I modified the userdata section, but the eez studio code output does not run and only a black screen is run and the code crashes.

mvladic, 2024-09-11:
> The output of the lv_event_get_user_data function is of pointer type and the compiler cannot cast it to an integer number.
my eeror:
cast from 'void*' to 'int32_t' {aka 'int'} loses precision [-fpermissive]

This is now fixed, so compiler will not complain when "-fpermissive" is enabled.

mvladic, 2024-09-11:
> Now I modified the userdata section, but the eez studio code output does not run and only a black screen is run and the code crashes.

How you modified?

elmonic, 2024-09-11:
I have not found anywhere to fix this problem.
..\ui\eez-flow.cpp: In function 'void flowPropagateValueLVGLEvent(void*, unsigned int, unsigned int, lv_event_t*)':
..\ui\eez-flow.cpp:6613:24: error: cast from 'void*' to 'int32_t' {aka 'int'} loses precision [-fpermissive]
 6613 |     int32_t userData = (int32_t)lv_event_get_user_data(event);
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

"How you modified?"
with by changing the variable to a pointer and using * to get its contents.

elmonic, 2024-09-15:
thanks
I am waiting for the next version, and to solve this issue, I use the eez library separately, and I disabled the option to add the eez library in the eez studio software.
In case, even if I modify it manually, after getting the output again, I have to use the correct code again, which I decided to use the same method as the first time.

closed 2024-09-28 by mvladic