eez-open/studio #532: Import images in bulk

HaydenHu, 2024-09-12:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

mvladic, 2024-09-12:
You can drag and drop multiple images from the file explorer to the studio like this:

[image]

mvladic, 2024-09-12:
I don't know if it helps. If not, please describe what solution you would like to have.

HaydenHu, 2024-09-12:
> I don't know if it helps. If not, please describe what solution you would like to have.

[image]
[image]
I can't select more image files here.,I thought I couldn't import in batches.,It's because I didn't find a way.,I'm wrong.

mvladic, 2024-09-12:
You are right, you can't select multiple images from the file open dialog - only way to add multiple images in bulk is by using drag and drop like I explained in my previous comment.

mvladic, 2024-09-12:
But, let's keep this issue open and I will allow to select multiple images in the New Bitmap dialog. When multiple images are selected then I will hide the Name field and name of the each bitmap added will be derived from the file name.

HaydenHu, 2024-09-12:
> 但是，让我们保持这个问题的开放性，我将允许在 New Bitmap 对话框中选择多个图像。当选择多个图像时，我将隐藏 Name 字段，并且添加的每个位图的名称将从文件名派生。

[image]
Here's how to do it in bulk?

mvladic, 2024-09-12:
Ok, I now understand what you want. This is not supported currently, but it is a valid request that should be implemented.

guochenwei-sp, 2024-11-05:
> You can drag and drop multiple images from the file explorer to the studio like this:
> 
> ![dd_images](https://private-user-images.githubusercontent.com/449788/366833434-02b4f4f1-6385-40d7-bb9c-0d7095586560.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA3OTkzNzcsIm5iZiI6MTczMDc5OTA3NywicGF0aCI6Ii80NDk3ODgvMzY2ODMzNDM0LTAyYjRmNGYxLTYzODUtNDBkNy1iYjljLTBkNzA5NTU4NjU2MC5naWY_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEwNVQwOTMxMTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yMDQ3NzYxMmUzNDNlNmQyZWQ1NTU2YWFkNWIzMjBlYzc1N2U0NzU1NDViYzkwNDVhZjlmNTI2N2ExZTRkZDg2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.sFJ86QACwzH3oNe2dA5rV9CL0ZbsrP0JdwCcnG18IAA) [ ![dd_images](https://private-user-images.githubusercontent.com/449788/366833434-02b4f4f1-6385-40d7-bb9c-0d7095586560.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA3OTkzNzcsIm5iZiI6MTczMDc5OTA3NywicGF0aCI6Ii80NDk3ODgvMzY2ODMzNDM0LTAyYjRmNGYxLTYzODUtNDBkNy1iYjljLTBkNzA5NTU4NjU2MC5naWY_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEwNVQwOTMxMTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yMDQ3NzYxMmUzNDNlNmQyZWQ1NTU2YWFkNWIzMjBlYzc1N2U0NzU1NDViYzkwNDVhZjlmNTI2N2ExZTRkZDg2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.sFJ86QACwzH3oNe2dA5rV9CL0ZbsrP0JdwCcnG18IAA) ](https://private-user-images.githubusercontent.com/449788/366833434-02b4f4f1-6385-40d7-bb9c-0d7095586560.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA3OTkzNzcsIm5iZiI6MTczMDc5OTA3NywicGF0aCI6Ii80NDk3ODgvMzY2ODMzNDM0LTAyYjRmNGYxLTYzODUtNDBkNy1iYjljLTBkNzA5NTU4NjU2MC5naWY_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEwNVQwOTMxMTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yMDQ3NzYxMmUzNDNlNmQyZWQ1NTU2YWFkNWIzMjBlYzc1N2U0NzU1NDViYzkwNDVhZjlmNTI2N2ExZTRkZDg2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.sFJ86QACwzH3oNe2dA5rV9CL0ZbsrP0JdwCcnG18IAA) [ ](https://private-user-images.githubusercontent.com/449788/366833434-02b4f4f1-6385-40d7-bb9c-0d7095586560.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA3OTkzNzcsIm5iZiI6MTczMDc5OTA3NywicGF0aCI6Ii80NDk3ODgvMzY2ODMzNDM0LTAyYjRmNGYxLTYzODUtNDBkNy1iYjljLTBkNzA5NTU4NjU2MC5naWY_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEwNVQwOTMxMTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yMDQ3NzYxMmUzNDNlNmQyZWQ1NTU2YWFkNWIzMjBlYzc1N2U0NzU1NDViYzkwNDVhZjlmNTI2N2ExZTRkZDg2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.sFJ86QACwzH3oNe2dA5rV9CL0ZbsrP0JdwCcnG18IAA)

I followed this method and was unable to import the png image, it still shows blank after mouse release.

mvladic, 2024-11-17:
> > You can drag and drop multiple images from the file explorer to the studio like this:
>
> I followed this method and was unable to import the png image, it still shows blank after mouse release.

I just found out that this feature stopped to work in version 0.17.0 through 0.18.2, but it is working again on version 0.19.0

mvladic, 2024-11-17:
> But, let's keep this issue open and I will allow to select multiple images in the New Bitmap dialog. When multiple images are selected then I will hide the Name field and name of the each bitmap added will be derived from the file name.

This is now implemented.