eez-open/studio #534: [EEZ Studio] Label box background color does not work

gwiethaus, 2024-09-13:
**Bug Description**

Sorry for my English.

Below is the figure that shows that the label does not work with the background color selection.

The border color is working.

**Screenshots**

[image]

**Desktop:**
 - OS: Windows 10
 - LVGL 9.x
- EEZ Studio 0.16.1

mvladic, 2024-09-13:
Which state is selected in styles editor? It is not visible in the screenshot.

mvladic, 2024-09-13:
Opacity for the label is 0 by default.

[image]

gwiethaus, 2024-09-17:
> Background opacity for the label is 0 by default.
> 
> ![image](https://private-user-images.githubusercontent.com/449788/367361175-4e1da1e1-3dad-43c0-9bea-b0ce16b018d3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY1NzA4MjksIm5iZiI6MTcyNjU3MDUyOSwicGF0aCI6Ii80NDk3ODgvMzY3MzYxMTc1LTRlMWRhMWUxLTNkYWQtNDNjMC05YmVhLWIwY2UxNmIwMThkMy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkxN1QxMDU1MjlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wODFiMmM1N2VmNGJjMjZiMGYwZGIzYWM0YTY3ZGU4YmY4YzdlYWRmNWU0OTJmNGIyMTkwODczMDk5OTczMzc4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.chgip0EtRBvtmJj7c8xKOyp6ciMIZIxGZLfBmePur88)

Sorry for my mistake. I really forgot to enable opacity, otherwise the color would be transparent. I hope I can contribute even more to improving the software.

closed 2024-09-27 by mvladic