eez-open/studio #535: npm run build error

gxdung, 2024-09-13:
```packages/eez-studio-shared/db.ts(98,31): error TS2339: Property 'description' does not exist on type '{}'.
packages/eez-studio-shared/db.ts(148,27): error TS18047: 'versionRow' is possibly 'null'.
packages/eez-studio-shared/db.ts(148,38): error TS2339: Property 'version' does not exist on type '{}'.
packages/eez-studio-shared/db.ts(338,36): error TS18046: 'row' is of type 'unknown'.
packages/eez-studio-shared/db.ts(339,33): error TS18046: 'row' is of type 'unknown'.
packages/eez-studio-shared/db.ts(361,36): error TS18046: 'row' is of type 'unknown'.
packages/eez-studio-shared/db.ts(362,33): error TS18046: 'row' is of type 'unknown'.
packages/eez-studio-shared/db.ts(391,21): error TS2769: No overload matches this call.
  Overload 1 of 2, '(o: {}): string[]', gave the following error.
    Argument of type 'unknown' is not assignable to parameter of type '{}'.
  Overload 2 of 2, '(o: object): string[]', gave the following error.
    Argument of type 'unknown' is not assignable to parameter of type 'object'.
packages/eez-studio-shared/db.ts(405,43): error TS18046: 'sourceInstrument' is of type 'unknown'.
packages/eez-studio-shared/db.ts(410,32): error TS18046: 'sourceInstrument' is of type 'unknown'.
packages/eez-studio-shared/db.ts(435,52): error TS2769: No overload matches this call.
  Overload 1 of 2, '(o: {}): string[]', gave the following error.
    Argument of type 'unknown' is not assignable to parameter of type '{}'.
  Overload 2 of 2, '(o: object): string[]', gave the following error.
    Argument of type 'unknown' is not assignable to parameter of type 'object'.
packages/eez-studio-shared/db.ts(444,31): error TS18046: 'sourceSession' is of type 'unknown'.
packages/eez-studio-shared/db.ts(447,36): error TS18046: 'sourceSession' is of type 'unknown'.
packages/eez-studio-shared/db.ts(448,55): error TS2339: Property 'id' does not exist on type '{}'.
packages/eez-studio-shared/db.ts(461,47): error TS18046: 'sourceSession' is of type 'unknown'.
packages/eez-studio-shared/db.ts(466,36): error TS18046: 'sourceSession' is of type 'unknown'.
packages/eez-studio-shared/db.ts(479,53): error TS2769: No overload matches this call.
  Overload 1 of 2, '(o: {}): string[]', gave the following error.
    Argument of type 'unknown' is not assignable to parameter of type '{}'.
  Overload 2 of 2, '(o: object): string[]', gave the following error.
    Argument of type 'unknown' is not assignable to parameter of type 'object'.
packages/eez-studio-shared/db.ts(485,25): error TS18046: 'sourceShortcut' is of type 'unknown'.
packages/eez-studio-shared/db.ts(505,43): error TS18046: 'sourceShortcut' is of type 'unknown'.
packages/eez-studio-shared/db.ts(537,46): error TS2769: No overload matches this call.
  Overload 1 of 2, '(o: {}): string[]', gave the following error.
    Argument of type 'unknown' is not assignable to parameter of type '{}'.
  Overload 2 of 2, '(o: object): string[]', gave the following error.
    Argument of type 'unknown' is not assignable to parameter of type 'object'.
packages/eez-studio-shared/db.ts(557,36): error TS2488: Type 'unknown' must have a '[Symbol.iterator]()' method that returns an iterator.
packages/eez-studio-shared/db.ts(559,44): error TS18046: 'log' is of type 'unknown'.
packages/eez-studio-shared/db.ts(561,33): error TS18046: 'log' is of type 'unknown'.
packages/eez-studio-shared/db.ts(563,50): error TS18046: 'log' is of type 'unknown'.
packages/eez-studio-shared/db.ts(566,62): error TS18046: 'log' is of type 'unknown'.
packages/eez-studio-shared/db.ts(574,50): error TS18046: 'logsCountRow' is of type 'unknown'.
packages/eez-studio-shared/db.ts(646,27): error TS18046: 'sourceInstrument' is of type 'unknown'.
packages/eez-studio-shared/db.ts(690,21): error TS2769: No overload matches this call.
  Overload 1 of 2, '(o: {}): string[]', gave the following error.
    Argument of type 'unknown' is not assignable to parameter of type '{}'.
  Overload 2 of 2, '(o: object): string[]', gave the following error.
    Argument of type 'unknown' is not assignable to parameter of type 'object'.
packages/eez-studio-shared/db.ts(704,43): error TS18046: 'sourceInstrument' is of type 'unknown'.
packages/eez-studio-shared/db.ts(709,32): error TS18046: 'sourceInstrument' is of type 'unknown'.
packages/eez-studio-shared/db.ts(722,52): error TS2769: No overload matches this call.
  Overload 1 of 2, '(o: {}): string[]', gave the following error.
    Argument of type 'unknown' is not assignable to parameter of type '{}'.
  Overload 2 of 2, '(o: object): string[]', gave the following error.
    Argument of type 'unknown' is not assignable to parameter of type 'object'.
packages/eez-studio-shared/db.ts(731,31): error TS18046: 'sourceSession' is of type 'unknown'.
packages/eez-studio-shared/db.ts(734,36): error TS18046: 'sourceSession' is of type 'unknown'.
packages/eez-studio-shared/db.ts(735,55): error TS2339: Property 'id' does not exist on type '{}'.
packages/eez-studio-shared/db.ts(748,47): error TS18046: 'sourceSession' is of type 'unknown'.
packages/eez-studio-shared/db.ts(753,36): error TS18046: 'sourceSession' is of type 'unknown'.
packages/eez-studio-shared/db.ts(766,53): error TS2769: No overload matches this call.
  Overload 1 of 2, '(o: {}): string[]', gave the following error.
    Argument of type 'unknown' is not assignable to parameter of type '{}'.
  Overload 2 of 2, '(o: object): string[]', gave the following error.
    Argument of type 'unknown' is not assignable to parameter of type 'object'.
packages/eez-studio-shared/db.ts(772,25): error TS18046: 'sourceShortcut' is of type 'unknown'.
packages/eez-studio-shared/db.ts(792,43): error TS18046: 'sourceShortcut' is of type 'unknown'.
packages/eez-studio-shared/db.ts(818,46): error TS2769: No overload matches this call.
  Overload 1 of 2, '(o: {}): string[]', gave the following error.
    Argument of type 'unknown' is not assignable to parameter of type '{}'.
[... 28 lines cut ...]
packages/notebook/export.tsx(85,36): error TS2339: Property 'label' does not exist on type '{}'.
packages/notebook/export.tsx(86,36): error TS2339: Property 'idn' does not exist on type '{}'.
packages/notebook/export.tsx(88,55): error TS2339: Property 'instrumentExtensionId' does not exist on type '{}'.
packages/notebook/store.ts(122,51): error TS2339: Property 'id' does not exist on type '{}'.
packages/notebook/store.ts(147,30): error TS2339: Property 'id' does not exist on type '{}'.
packages/notebook/store.ts(150,28): error TS2339: Property 'instrumentExtensionId' does not exist on type '{}'.
packages/notebook/store.ts(151,28): error TS2339: Property 'label' does not exist on type '{}'.
packages/notebook/store.ts(152,28): error TS2339: Property 'idn' does not exist on type '{}'.
packages/notebook/store.ts(154,24): error TS2339: Property 'instrumentExtensionId' does not exist on type '{}'.
packages/notebook/store.ts(171,9): error TS2322: Type 'unknown' is not assignable to type 'INotebookItemSource | null'.
packages/project-editor/features/font/font-extract/opentype.ts(115,25): error TS18048: 'glyph.advanceWidth' is possibly 'undefined'.
packages/project-editor/lvgl/Page.tsx(70,18): error TS2741: Property 'forwardedRef' is missing in type '{ ref: RefObject<HTMLCanvasElement>; width: number; height: number; flowContext: IFlowContext; }' but required in type 'LVGLPageCanvasProps'.
packages/project-editor/ui-components/Tree.tsx(383,42): error TS2532: Object is possibly 'undefined'.
packages/project-editor/ui-components/Tree.tsx(397,41): error TS2532: Object is possibly 'undefined'.
packages/project-editor/ui-components/Tree.tsx(400,43): error TS2532: Object is possibly 'undefined'.
packages/project-editor/ui-components/Tree.tsx(499,63): error TS2532: Object is possibly 'undefined'.
packages/project-editor/ui-components/Tree.tsx(562,51): error TS2532: Object is possibly 'undefined'.
packages/project-editor/ui-components/Tree.tsx(598,62): error TS2532: Object is possibly 'undefined'.
packages/project-editor/ui-components/Tree.tsx(758,38): error TS2769: No overload matches this call.
  The last overload gave the following error.
    Argument of type 'HTMLElement | undefined' is not assignable to parameter of type 'PlainObject<any>'.
      Type 'undefined' is not assignable to type 'PlainObject<any>'.
packages/project-editor/ui-components/Tree.tsx(758,52): error TS2339: Property 'attr' does not exist on type 'JQueryStatic'.

```

mvladic, 2024-09-13:
This is probably problem with the wrong version of TypeScript compiler. If you have TypeScript compiler installed globally, make sure it is version 5.5.3. You can check your TypeScript compiler version inside studio directory by running `npm run tsc-version`:

[image]

gxdung, 2024-09-15:
I encountered other issues, please assist.

```
PS D:\Workspace\EEZStudio-Source\studio> npm install
⠸
PS D:\Workspace\EEZStudio-Source\studio> ^C
PS D:\Workspace\EEZStudio-Source\studio> ^C
PS D:\Workspace\EEZStudio-Source\studio> npm install
npm warn deprecated @types/classnames@2.3.1: This is a stub types definition. classnames provides its own type definitions, so you do not need this installed.
npm warn deprecated inflight@1.0.6: This module is not supported, and leaks memory. Do not use it. Check out lru-cache if you want a good and tested way to coalesce async requests by a key value, which is much more comprehensive and powerful.
npm warn deprecated request-promise-native@0.0.0: request-promise-native has been deprecated because it extends the now deprecated request package, see https://github.com/request/request/issues/3142
npm warn deprecated @npmcli/move-file@2.0.1: This functionality has been moved to @npmcli/fs
npm warn deprecated npmlog@6.0.2: This package is no longer supported.
npm warn deprecated flatten@1.0.3: flatten is deprecated in favor of utility frameworks such as lodash.
npm warn deprecated source-map-resolve@0.6.0: See https://github.com/lydell/source-map-resolve#deprecated
npm warn deprecated rimraf@2.7.1: Rimraf versions prior to v4 are no longer supported
npm warn deprecated har-validator@5.1.5: this library is no longer supported
npm warn deprecated rimraf@3.0.2: Rimraf versions prior to v4 are no longer supported
npm warn deprecated glob@7.2.3: Glob versions prior to v9 are no longer supported
npm warn deprecated glob@8.1.0: Glob versions prior to v9 are no longer supported
npm warn deprecated abab@2.0.6: Use your platform's native atob() and btoa() methods instead
npm warn deprecated are-we-there-yet@3.0.1: This package is no longer supported.
npm warn deprecated formidable@1.2.6: Please upgrade to latest, formidable@v2 or formidable@v3! Check these notes: https://bit.ly/2ZEqIau
npm warn deprecated domexception@4.0.0: Use your platform's native DOMException instead
npm warn deprecated xterm-addon-fit@0.5.0: This package is now deprecated. Move to @xterm/addon-fit instead.
npm warn deprecated gauge@4.0.4: This package is no longer supported.
npm warn deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
npm warn deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm warn deprecated superagent@7.1.6: Please upgrade to v9.0.0+ as we have fixed a public vulnerability with formidable dependency. Note that v9.0.0+ requires Node.js v14.18.0+. See https://github.com/ladjs/superagent/pull/1800 for insight. This project is supported and maintained by the team at Forward Email @ https://forwardemail.net
npm warn deprecated superagent@4.1.0: Please upgrade to v9.0.0+ as we have fixed a public vulnerability with formidable dependency. Note that v9.0.0+ requires Node.js v14.18.0+. See https://github.com/ladjs/superagent/pull/1800 for insight. This project is supported and maintained by the team at Forward Email @ https://forwardemail.net
npm warn deprecated electron-rebuild@3.2.9: Please use @electron/rebuild moving forward.  There is no API change, just a package name change
npm warn deprecated xterm@4.19.0: This package is now deprecated. Move to @xterm/xterm instead.
npm error code 1
npm error path D:\Workspace\EEZStudio-Source\studio\node_modules\better-sqlite3
npm error command failed
npm error command C:\Windows\system32\cmd.exe /d /s /c prebuild-install || node-gyp rebuild --release
npm error prebuild-install warn install No prebuilt binaries found (target=22.4.0 runtime=node arch=x64 libc= platform=win32)
npm error gyp info it worked if it ends with ok
npm error gyp info using node-gyp@9.3.1
npm error gyp info using node@22.4.0 | win32 | x64
npm error (node:66276) [DEP0060] DeprecationWarning: The `util._extend` API is deprecated. Please use Object.assign() instead.
npm error (Use `node --trace-deprecation ...` to show where the warning was created)
npm error gyp info find Python using Python version 3.12.6 found at "C:\Python312\python.exe"
npm error gyp info find VS using VS2022 (17.11.35303.130) found at:
npm error gyp info find VS "C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools"
npm error gyp info find VS run with --verbose for detailed information
npm error gyp info spawn C:\Python312\python.exe
npm error gyp info spawn args [
npm error gyp info spawn args   'D:\\Workspace\\EEZStudio-Source\\studio\\node_modules\\node-gyp\\gyp\\gyp_main.py',
npm error gyp info spawn args   'binding.gyp',
npm error gyp info spawn args   '-f',
npm error gyp info spawn args   'msvs',
npm error gyp info spawn args   '-I',
npm error gyp info spawn args   'D:\\Workspace\\EEZStudio-Source\\studio\\node_modules\\better-sqlite3\\build\\config.gypi',
npm error gyp info spawn args   '-I',
npm error gyp info spawn args   'D:\\Workspace\\EEZStudio-Source\\studio\\node_modules\\node-gyp\\addon.gypi',
npm error gyp info spawn args   '-I',
npm error gyp info spawn args   'C:\\Users\\gxdung\\AppData\\Local\\node-gyp\\Cache\\22.4.0\\include\\node\\common.gypi',
npm error gyp info spawn args   '-Dlibrary=shared_library',
npm error gyp info spawn args   '-Dvisibility=default',
[... cut ...]
npm error gyp ERR! System Windows_NT 10.0.22631
npm error gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "D:\\Workspace\\EEZStudio-Source\\studio\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild" "--release"
npm error gyp ERR! cwd D:\Workspace\EEZStudio-Source\studio\node_modules\better-sqlite3
npm error gyp ERR! node -v v22.4.0
npm error gyp ERR! node-gyp -v v9.3.1
npm error gyp ERR! not ok
npm error A complete log of this run can be found in: C:\Users\gxdung\AppData\Local\npm-cache\_logs\2024-09-15T07_09_47_708Z-debug-0.log
```

closed 2024-10-17 by mvladic