eez-open/studio #536: Grid flow please?

joysfera, 2024-09-14:
The LVGL's Grid flow is missing.

[https://docs.lvgl.io/master/layouts/grid.html](https://docs.lvgl.io/master/layouts/grid.html)

I'd like to create a grid of identical widgets on the screen/display.

mvladic, 2024-09-27:
Implemented!

arbymohamed, 2025-01-16:
hello, i tried using this grid layout but i cant make it work. All the widget are just in one corner of the container i put them in.
us there a guide for this tool?

mvladic, 2025-01-16:
By default, a Widget's children are not added to the grid. They need to be added manually to a cell.

[image]

Here is an example project:

[lvgl_grid_test.zip](https://github.com/user-attachments/files/18444696/lvgl_grid_test.zip)

arbymohamed, 2025-01-16:
thanks for the details, it is working now. Do all the widgets have to be the same? Let's say i want to arrange lables and drop_downs, would that work?

arbymohamed, 2025-01-16:
[image]

i tried to adda drop down here, but its now in thre right position. If you look at the picture attached, the drop-dowb should be in col 0 row 2, right below row 0 and row 1 where the labels are. Maybe i did somethign wrong

mvladic, 2025-01-16:
You miss configured something, maybe"Grid row descriptor" of the Panel widget, it should have three numbers since you have three rows. Check my example:

[lvgl_grid_test2.zip](https://github.com/user-attachments/files/18445354/lvgl_grid_test2.zip)

[image]

arbymohamed, 2025-01-16:
it was my fault, lol. i was modifying the selected part of the drop-down all this time. sorry if i waisted your time.
every works very well now. thank you

AwangYes, 2026-02-03:
> > Implemented!   实现!
> > Note: "Grid row descriptor" and "Grid column descriptor" is specified as a list of number, separated by space or comma:注意：“网格行描述符”和“网格列描述符”被指定为由空格或逗号分隔的数字列表：
> > [image]
> 
> So LV_GRID_FR and LV_GRID_CONTENT are still not supported?所以 LV_GRID_FR 和 LV_GRID_CONTENT 仍然不受支持吗？

@mvladic Dear, I hope you can consider supporting the LV_GRID_FR and LV_GRID_CONTENT attributes. Thank you!