eez-open/studio #537: EZZ Not starting on mac os 15

Obili69, 2024-09-14:
**Describe the bug**
EZZ Studio doesnt start on Macos 15.0

**To Reproduce**
EZZ crashes on start. no window appearing

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
Macos 15.0 on Macbook pro m1

** LVGL version (if used)**

- 8.x or 9.x

**Additional context**
Add any other context about the problem here.

ddiogo22, 2024-09-17:
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               EEZ Studio [22118]
Path:                  /Users/USER/*/EEZ Studio.app/Contents/MacOS/EEZ Studio
Identifier:            hr.envox.eez.studio
Version:               0.16.1 (0.16.1)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               503

Date/Time:             2024-09-17 05:00:01.9105 +0100
OS Version:            macOS 15.0 (24A335)
Report Version:        12
Anonymous UUID:        663AFD46-14B5-DA8B-DCDF-7393BB316E6F

Time Awake Since Boot: 4100 seconds

System Integrity Protection: disabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000001, 0x000000010bc39d38

Termination Reason:    Namespace SIGNAL, Code 5 Trace/BPT trap: 5
Terminating Process:   exc handler [22118]

Thread 0 Crashed:: CrBrowserMain Dispatch queue: com.apple.main-thread
0   Electron Framework            	       0x10bc39d38 electron::fuses::IsLoadBrowserProcessSpecificV8SnapshotEnabled() + 45844
1   Electron Framework            	       0x10bc39d38 electron::fuses::IsLoadBrowserProcessSpecificV8SnapshotEnabled() + 45844
2   Electron Framework            	       0x10bc4f6d4 v8::internal::GetCurrentStackPosition() + 35436
3   libswiftCore.dylib            	       0x1a17f3a10 _swift_release_dealloc + 56
4   libswiftCore.dylib            	       0x1a17f4a64 bool swift::RefCounts<swift::RefCountBitsT<(swift::RefCountInlinedness)1>>::doDecrementSlow<(swift::PerformDeinit)1>(swift::RefCountBitsT<(swift::RefCountInlinedness)1>, unsigned int) + 160
5   libobjc.A.dylib               	       0x190447d94 AutoreleasePoolPage::releaseUntil(objc_object**) + 204
6   libobjc.A.dylib               	       0x190444138 objc_autoreleasePoolPop + 260
7   CoreFoundation                	       0x1908b87c0 _CFAutoreleasePoolPop + 32
8   AppKit                        	       0x19442f3c0 -[NSApplication finishLaunching] + 1772
9   AppKit                        	       0x19442ea98 -[NSApplication run] + 252
10  Electron Framework            	       0x10eda3318 node::GetArrayBufferAllocator(node::IsolateData*) + 17973960
11  Electron Framework            	       0x10eda1184 node::GetArrayBufferAllocator(node::IsolateData*) + 17965364
12  Electron Framework            	       0x10ed59ed4 node::GetArrayBufferAllocator(node::IsolateData*) + 17673860
13  Electron Framework            	       0x10ed20abc node::GetArrayBufferAllocator(node::IsolateData*) + 17439340
14  Electron Framework            	       0x10deb3164 node::GetArrayBufferAllocator(node::IsolateData*) + 2310420
15  Electron Framework            	       0x10deb4bfc node::GetArrayBufferAllocator(node::IsolateData*) + 2317228
16  Electron Framework            	       0x10deb0ad4 node::GetArrayBufferAllocator(node::IsolateData*) + 2300548
17  Electron Framework            	       0x10bd4d288 v8::internal::compiler::BasicBlock::set_loop_header(v8::internal::compiler::BasicBlock*) + 13948
18  Electron Framework            	       0x10bd4e3f0 v8::internal::compiler::BasicBlock::set_loop_header(v8::internal::compiler::BasicBlock*) + 18404
19  Electron Framework            	       0x10bd4e220 v8::internal::compiler::BasicBlock::set_loop_header(v8::internal::compiler::BasicBlock*) + 17940
20  Electron Framework            	       0x10bd4cacc v8::internal::compiler::BasicBlock::set_loop_header(v8::internal::compiler::BasicBlock*) + 11968
21  Electron Framework            	       0x10bd4cc6c v8::internal::compiler::BasicBlock::set_loop_header(v8::internal::compiler::BasicBlock*) + 12384
22  Electron Framework            	       0x10ba1e2f0 ElectronMain + 128
23  dyld                          	       0x190490274 start + 2840

Thread 1:
0   libsystem_pthread.dylib       	       0x19080e0e8 start_wqthread + 0

Thread 2:
0   libsystem_pthread.dylib       	       0x19080e0e8 start_wqthread + 0

Thread 3:
0   libsystem_pthread.dylib       	       0x19080e0e8 start_wqthread + 0

Thread 4:: ThreadPoolServiceThread
0   libsystem_kernel.dylib        	       0x1907ddbdc kevent64 + 8
1   Electron Framework            	       0x10edace20 node::GetArrayBufferAllocator(node::IsolateData*) + 18013648
2   Electron Framework            	       0x10edac654 node::GetArrayBufferAllocator(node::IsolateData*) + 18011652
3   Electron Framework            	       0x10ed59ed4 node::GetArrayBufferAllocator(node::IsolateData*) + 17673860
4   Electron Framework            	       0x10ed20abc node::GetArrayBufferAllocator(node::IsolateData*) + 17439340
[... 623 lines cut ...]
      },
      "deploymentId" : 240000004
    },
    {
      "rolloutId" : "648a2601f74c42732876cb5b",
      "factorPackIds" : {
        "SIRI_TEXT_TO_SPEECH" : "66d7b9d1d9aabf7ca9408524"
      },
      "deploymentId" : 240000244
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "c47ab4cc-c9c3-4b5d-a87c-e2433ce02597",
      "experimentId" : "6639bc6ba73d460582162323",
      "deploymentId" : 400000006
    },
    {
      "treatmentId" : "45f4e2a5-551b-4bc2-a2dc-19c244dda8f8",
      "experimentId" : "6643969b3099cf28e049862f",
      "deploymentId" : 400000007
    }
  ]
}
}

mvladic, 2024-09-17:
I upgraded some dependencies (mainly [Electron](https://www.electronjs.org/) to the latest version) and now EEZ Studio works on MacOS 15.

Now, I need to do more tests on all supported platforms (MacOS, Windows and Linux).

The next release (version 0.17.0) should be in two weeks.

Obili69, 2024-09-17:
Thanks a lot

BoraMD, 2024-09-25:
EEZ.Studio-0.16.1.dmg on MacBook Pro 2019 16" crashes same. Here's the report.
`-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               EEZ Studio [26029]
Path:                  /Applications/EEZ Studio.app/Contents/MacOS/EEZ Studio
Identifier:            hr.envox.eez.studio
Version:               0.16.1 (0.16.1)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-09-25 14:47:09.7762 +0300
OS Version:            macOS 15.0 (24A335)
Report Version:        12
Bridge OS Version:     9.0 (22P353)
Anonymous UUID:        B17BA532-5321-5E10-F305-490B18F806A6

Sleep/Wake UUID:       FF4D68D6-E21A-4F06-B763-EDF0603974AC

Time Awake Since Boot: 100000 seconds
Time Since Wake:       833 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000002, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 5 Trace/BPT trap: 5
Terminating Process:   exc handler [26029]

Thread 0 Crashed:: CrBrowserMain Dispatch queue: com.apple.main-thread
0   Electron Framework            	       0x111425030 electron::fuses::IsLoadBrowserProcessSpecificV8SnapshotEnabled() + 47600
1   Electron Framework            	       0x11143ece9 v8::internal::GetCurrentStackPosition() + 40057
2   libcache.dylib                	    0x7ff81ddfde90 _value_entry_table_resize + 596
3   libcache.dylib                	    0x7ff81ddfe117 cache_set_and_retain + 525
4   CoreFoundation                	    0x7ff80feb875f -[NSCache setObject:forKey:cost:] + 405
5   CoreText                      	    0x7ff8124b347b TCGFontCache::CopyFont(__CFURL const*, __CFString const*, __CFString const*) + 195
6   CoreText                      	    0x7ff81237d807 TBaseFont::CopyNativeFont() const + 143
7   CoreText                      	    0x7ff8124a60b8 TTenuousComponentFont::InitializeVariationGraphicsFont() const + 72
8   CoreText                      	    0x7ff812386e56 TTenuousComponentFont::GetGraphicsFont() const + 44
9   CoreText                      	    0x7ff8123ec2fe TBaseFont::GetInitializedGraphicsFont() const + 22
10  CoreText                      	    0x7ff8124a068b TBaseFont::CalculateFontMetrics(bool, double) const + 37
11  CoreText                      	    0x7ff8124a463d TComponentFont::CalculateFontMetrics(bool, double) const + 97
12  CoreText                      	    0x7ff8124a0b5e TBaseFont::InitFontMetrics(double) const + 104
13  CoreText                      	    0x7ff81237d62f TBaseFont::GetStrikeMetrics(double, CGAffineTransform const*, bool) const + 177
14  CoreText                      	    0x7ff812429286 TFont::GetStrikeMetrics(bool) const + 188
15  CoreText                      	    0x7ff81237d3b4 TFont::InitStrikeMetrics() const + 34
16  CoreText                      	    0x7ff81237d384 CTFontGetAscent + 34
17  UIFoundation                  	    0x7ff814c7c20f -[__NSFontExtraData initWithFont:] + 108
18  UIFoundation                  	    0x7ff814d533c8 __CTFontGetExtraData + 49
19  UIFoundation                  	    0x7ff814c7c13d -[NSFont initWithTypefaceInfo:key:renderingMode:] + 47
20  UIFoundation                  	    0x7ff814c7bf70 -[__NSFontTypefaceInfo fontInstanceForKey:renderingMode:] + 100
21  UIFoundation                  	    0x7ff814c7beec -[__NSFontTypefaceInfo fontInstanceForFontDescriptor:size:affineTransform:renderingMode:] + 335
22  UIFoundation                  	    0x7ff814c7aff6 __NSGetMetaFontInstanceWithType + 606
23  AppKit                        	    0x7ff81418dbe7 -[NSMenuBarItemView _effectiveFontRespectingPreferredParameters:] + 226
24  AppKit                        	    0x7ff81418cb56 -[NSMenuBarItemView _ensureValidLineCache] + 144
25  AppKit                        	    0x7ff81418d7ad -[NSMenuBarItemView idealWidth] + 14
26  AppKit                        	    0x7ff8140c1933 -[NSMenuBarRepresentation _menuItemViewWidths] + 619
27  AppKit                        	    0x7ff8140bf567 __48-[NSMenuBarRepresentation layoutMenuBarIfNeeded]_block_invoke + 1229
28  AppKit                        	    0x7ff813866058 NSPerformVisuallyAtomicChange + 134
29  AppKit                        	    0x7ff8140bf064 -[NSMenuBarRepresentation layoutMenuBarIfNeeded] + 245
30  AppKit                        	    0x7ff8146eb846 __46-[NSMenuBarImpl layoutAllMenuBarsImmediately:]_block_invoke + 57
31  AppKit                        	    0x7ff8146e83a6 -[NSMenuBarImpl forEachRepresentationViewDo:] + 213
32  AppKit                        	    0x7ff8146eb804 -[NSMenuBarImpl layoutAllMenuBarsImmediately:] + 192
33  AppKit                        	    0x7ff8146d546c -[NSMenuBarPresentationInstance _setBoundsAndUpdateResolution] + 130
34  AppKit                        	    0x7ff8146d310d -[NSMenuBarPresentationInstance _menuImplDidChange] + 103
[... 737 lines cut ...]

Model: MacBookPro16,1, BootROM 2069.0.0.0.0 (iBridge: 22.16.10353.0.0,0), 6 processors, 6-Core Intel Core i7, 2,6 GHz, 16 GB, SMC 
Graphics: Intel UHD Graphics 630, Intel UHD Graphics 630, Built-In
Graphics: AMD Radeon Pro 5300M, AMD Radeon Pro 5300M, PCIe, 4 GB
Display: Color LCD, 3072 x 1920 Retina, Main, MirrorOff, Online
Memory Module: BANK 0/ChannelA-DIMM0, 8 GB, DDR4, 2667 MHz, Micron, 8ATF1G64HZ-2G6E1
[... cut ...]
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: Apple Watch Magnetic Charging Cable
USB Device: T2Bus
USB Device: composite_device
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5
`

mvladic, 2024-09-25:
Version 0.17.0 which will be released by Oct 1 will work on MacOS 15.

closed 2024-09-27 by prasimix

Obili69, 2024-09-30:
> Studio 0.17.0 is released. Is it working for you now on macOS 15?

Thanks a lot its working now