eez-open/studio #544: Application crashing when editing LVGL Dropdown Menu widget with screen applied style.

lylezumbrun, 2024-09-18:
Application crashes when editing a dropdown menu widget size has been added to a page with applied style.

**To Reproduce**
Steps to reproduce the behavior:
1. Add page and apply a screen style with the background and text color set
2. Add a drop down menu and adjust the size of the drop down menu

**Screenshots**

**Desktop (please complete the following information):**
  Operating System: Linux Mint 20.1
  Kernel: Linux 5.4.0-167-generic
 Architecture: x86-64

** LVGL version (if used)**

- 8.x

lylezumbrun, 2024-09-19:
I have attached a video showing the error.

https://github.com/user-attachments/assets/a4e07e0e-2ebb-4057-b3ba-7e4999037fc3

mvladic, 2024-09-19:
We are doing almost the same thing, so something else in your project is causing the crash. Can I see all style modifications in your "Dark Screen" style? Or, if you can send me your project so I can investigate more directly.

lylezumbrun, 2024-09-19:
I have uploaded my project here:
[EEZ-Display-master.zip](https://github.com/user-attachments/files/17060705/EEZ-Display-master.zip)

mvladic, 2024-09-19:
Thanks! Now I was able to reproduce the problem.

lylezumbrun, 2024-09-19:
Thanks for you hard work! This software is really helpful!!!

mvladic, 2024-09-19:
This is fixed now in the master branch. FYI Next version 0.17.0 will be released by Oct 1.

closed 2024-09-26 by mvladic