eez-open/studio #552: User actions with input cause error in simulator upon start

fietser28, 2024-09-25:
**Describe the bug**
When you have user widget with an input the simulator will stop the flow with the error 'No parentFlowState in Input'. Even if the user action is not used in any screen!
It seems the user action is also started like a screen. 
This was introduced very recently. Maybe with the addition of keyboard inputs?

**To Reproduce**
Steps to reproduce the behavior:
0. Using latest commit from GitHub. 
1. Open the  attached project file. Note: the user widget IS NOT USED.
[simulatorBug.eez-project.zip](https://github.com/user-attachments/files/17138608/simulatorBug.eez-project.zip)
2. Start the debugger.
3. Look at the queue panel: myWidget items are queued, they shouldn't
4. Look at the Active flows panel: myWidget is shown as an active flow, it shouldn't.
5. Step thru.
6. When the input of the widget is executed the error is triggered.

**Expected behavior**
user widget shouldn't get executed by itself, only as child of a screen flow.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Mac
 - Version 14.5

** LVGL version (if used)**
-  9.x

mvladic, 2024-09-26:
Fixed.

fietser28, 2024-09-26:
Working, thank you!

closed 2024-09-26 by fietser28