eez-open/studio #553: [LVGL with Flow] Missing component in the list (Target Type) of the LVGL Action property

gwiethaus, 2024-09-26:
I believe that the following components are missing from the "target type" property: TextArea, Checkbox, Button, among others. This is indicated in the figure below.

Version: EEZ Studio 0.16.1

[image]

mvladic, 2024-09-26:
Currently, LVGL action supports only limited set of properties. If some widget type is not listed in target type then you can only use "Basic" target type and in that case only following properties are available:

[image]

gwiethaus, 2024-09-26:
Yes, it has now appeared for me. I think there was a problem updating the data.

In the property field, as you show in the figure, I wonder if it wouldn't be interesting to have the Text property of the component for the label and textarea?

Sorry for my previous mistake. I'm trying to help to further improve EEZ Studio.

mvladic, 2024-09-26:
For the Text property of the Label and Textarea widgets there is an alternative way to update its value:

[image]

If variables are used in the expression (such as `user_name` in this example) then the text of the label will be updated when that variable changes its value during execution. Updating is done inside `ui_tick` function, so for this to work you need to call periodically `ui_tick()`, for example after `lv_task_handler()`.

mvladic, 2024-09-26:
Simple example of using variables and expressions:

[counter.zip](https://github.com/user-attachments/files/17149196/counter.zip)

[image]

[image]

gwiethaus, 2024-09-26:
Thank you very much @mvladic. That's what I was thinking. So I can only do this update externally using the code directly in LVGL in the loop().

gwiethaus, 2024-09-26:
> Simple example of using variables and expressions:
> 
> [counter.zip](https://github.com/user-attachments/files/17149196/counter.zip)
> 
> ![image](https://private-user-images.githubusercontent.com/449788/371162366-f37d2995-9644-4444-8a3a-1135e2117067.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjczNTc1ODcsIm5iZiI6MTcyNzM1NzI4NywicGF0aCI6Ii80NDk3ODgvMzcxMTYyMzY2LWYzN2QyOTk1LTk2NDQtNDQ0NC04YTNhLTExMzVlMjExNzA2Ny5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyNlQxMzI4MDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hNGEzMTUyMDlmNmZkY2IyNDFmODY3ZTkzNGRiYmNjODAzZTZkMzI4YmZlODc2ZDhiZmNmNmJiZWE5ZjAyYTQyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.0ZYUIohm4_IH1lal4GCgBkNtNViWNJ3TkNEvFHKxPjo)
> 
> ![counter](https://private-user-images.githubusercontent.com/449788/371162757-d23dbe18-7205-457f-beb8-d6583e7789c2.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjczNTc1ODcsIm5iZiI6MTcyNzM1NzI4NywicGF0aCI6Ii80NDk3ODgvMzcxMTYyNzU3LWQyM2RiZTE4LTcyMDUtNDU3Zi1iZWI4LWQ2NTgzZTc3ODljMi5naWY_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyNlQxMzI4MDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jNzI5NjE5ZGMwYjQwN2MxMGM4NmQ5YjE3OWJkNTY3OWY2MGE1OGE3MzUwZTMyMGNkNjA3OTY1OTU0ZjQ4OTUzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.t0NceHMh55DHabOwQT9NRNFSvjVPPVdICF0nG-r0Lmk) [ ![counter](https://private-user-images.githubusercontent.com/449788/371162757-d23dbe18-7205-457f-beb8-d6583e7789c2.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjczNTc1ODcsIm5iZiI6MTcyNzM1NzI4NywicGF0aCI6Ii80NDk3ODgvMzcxMTYyNzU3LWQyM2RiZTE4LTcyMDUtNDU3Zi1iZWI4LWQ2NTgzZTc3ODljMi5naWY_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyNlQxMzI4MDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jNzI5NjE5ZGMwYjQwN2MxMGM4NmQ5YjE3OWJkNTY3OWY2MGE1OGE3MzUwZTMyMGNkNjA3OTY1OTU0ZjQ4OTUzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.t0NceHMh55DHabOwQT9NRNFSvjVPPVdICF0nG-r0Lmk) ](https://private-user-images.githubusercontent.com/449788/371162757-d23dbe18-7205-457f-beb8-d6583e7789c2.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjczNTc1ODcsIm5iZiI6MTcyNzM1NzI4NywicGF0aCI6Ii80NDk3ODgvMzcxMTYyNzU3LWQyM2RiZTE4LTcyMDUtNDU3Zi1iZWI4LWQ2NTgzZTc3ODljMi5naWY_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyNlQxMzI4MDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jNzI5NjE5ZGMwYjQwN2MxMGM4NmQ5YjE3OWJkNTY3OWY2MGE1OGE3MzUwZTMyMGNkNjA3OTY1OTU0ZjQ4OTUzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.t0NceHMh55DHabOwQT9NRNFSvjVPPVdICF0nG-r0Lmk) [ ](https://private-user-images.githubusercontent.com/449788/371162757-d23dbe18-7205-457f-beb8-d6583e7789c2.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjczNTc1ODcsIm5iZiI6MTcyNzM1NzI4NywicGF0aCI6Ii80NDk3ODgvMzcxMTYyNzU3LWQyM2RiZTE4LTcyMDUtNDU3Zi1iZWI4LWQ2NTgzZTc3ODljMi5naWY_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyNlQxMzI4MDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jNzI5NjE5ZGMwYjQwN2MxMGM4NmQ5YjE3OWJkNTY3OWY2MGE1OGE3MzUwZTMyMGNkNjA3OTY1OTU0ZjQ4OTUzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.t0NceHMh55DHabOwQT9NRNFSvjVPPVdICF0nG-r0Lmk)

Taking advantage of this. Could you give me a tip on how I can enable/disable a button according to the textarea filling condition, for example when I type 2 characters the button is enabled. I haven't been able to do this yet. Can you help me?

mvladic, 2024-09-26:
> So I can only do this update externally using the code directly in LVGL in the loop().

Yes. Studio generates code for you. You only need to call `ui_init()` and `ui_tick()`. For example on Arduino it will look something like this:

```
#include "ui.h" // this header file is generated by the studio when you press the Build icon

void setup() {
   ...
   lv_init();
   ui_init();
   ...
}

void setup() {
   ...
   lv_task_handler();
   ui_tick();
   ...
}

```

gwiethaus, 2024-09-26:
> > Taking advantage of this. Could you give me a tip on how I can enable/disable a button according to the textarea filling condition, for example when I type 2 characters the button is enabled. I haven't been able to do this yet. Can you help me?
> 
> Here it is:
> 
> [enable_button.zip](https://github.com/user-attachments/files/17149379/enable_button.zip)
> 
> ![image](https://private-user-images.githubusercontent.com/449788/371167408-b54a52da-e87a-4baf-8180-899bbb13ae5a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjczNTk4MTUsIm5iZiI6MTcyNzM1OTUxNSwicGF0aCI6Ii80NDk3ODgvMzcxMTY3NDA4LWI1NGE1MmRhLWU4N2EtNGJhZi04MTgwLTg5OWJiYjEzYWU1YS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyNlQxNDA1MTVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iYTA3YzZiZGU0YmZkMzc0YzUxYTUyMDA1NTc0YTAwMmU0N2ZjMDdhYjVhNGE0MDA4MGVjMmYzMmFlODE4NWM3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.rs8yPekx7gb899K_ZRYBz6iC-WSjOIWrQmiyONj0xx4)
> 
> If you have more questions, please ask on our [Discord](https://discord.gg/XxsNUNEt) server.

LOL... This makes it so easy! Thank you very much again. These things may seem silly, but for those who are just starting to explore EEZ Studio, it can be hellish at first.

I am understanding many questions about how to use it. I have already learned a lot just by practicing and experimenting. I want to teach people in a group here in Brazil (exchange of ideas and technologies - FabLab) how to use this excellent tool.

I didn't know about this discord channel to ask questions.

Thank you very much again for your help. If I find any errors in the software or any other ideas for improvement, I will be sure to suggest them.

closed 2024-09-27 by mvladic