eez-open/studio #554: [LVGL with Flow]-want to access Global variables

DhyeyBhavsar, 2024-09-27:
I have defined many Global Variables and that variable I want to use in main.ino file but not able to as I dont know how to get that values. Pls help me to get access for the variables.

For example:-
enum FlowGlobalVariables {
    FLOW_GLOBAL_VARIABLE_COUNT = 0,};
this the variable i want to use in arduino main.ino file

mvladic, 2024-09-27:
To get variable value:

```
#include "ui.h"
#include "vars.h"

int32_t count = eez::flow::getGLobalVariable(FLOW_GLOBAL_VARIABLE_COUNT).getInt();
const char *str_var = eez::flow::getGLobalVariable(FLOW_GLOBAL_VARIABLE_STR_VAR).getString();
bool bool_var = eez::flow::getGLobalVariable(FLOW_GLOBAL_VARIABLE_BOOL_VAR).getBool();
```

To set variable value:

```
#include "ui.h"
#include "vars.h"

using namespace eez;
using namespace eez::flow;

setGLobalVariable(FLOW_GLOBAL_VARIABLE_COUNT, Value(count, VALUE_TYPE_INT32));
setGLobalVariable(FLOW_GLOBAL_VARIABLE_STR_VAR, Value::MakeStringRef(str_var, 0));
setGLobalVariable(FLOW_GLOBAL_VARIABLE_BOOL_VAR, Value(bool_var, VALUE_TYPE_BOOL));
```

For arrays and structures check: https://github.com/eez-open/native-interface-lvgl-with-flow/blob/master/src/ui/vars.cpp

If you have more question you can ask on our [Discord server](https://discord.gg/5PP7s8xD).

closed 2024-09-27 by mvladic

DhyeyBhavsar, 2024-09-27:
Thank you so much!! 

It worked just for someone else there is typing error Global is mistakenly written as GLobal:-

Incorrect:
int32_t count = eez::flow::getGLobalVariable(FLOW_GLOBAL_VARIABLE_COUNT).getInt();

Corrected:
int32_t count = eez::flow::getGlobalVariable(FLOW_GLOBAL_VARIABLE_COUNT).getInt();