eez-open/studio #556: After Update Macos M1 Sequoia 15.0 - EEZ Studio 16.1 App Crash!

LEKPCSTEAM, 2024-09-28:
**Screenshots**
[image]

[image]

**Desktop (please complete the following information):**

 - OS: MacOS M1
 - Version : Sequoia 15.0

** LVGL version (if used)**

- 8.x or 9.x

**Report**
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               EEZ Studio [46692]
Path:                  /Applications/EEZ Studio.app/Contents/MacOS/EEZ Studio
Identifier:            hr.envox.eez.studio
Version:               0.15.1 (0.15.1)
Code Type:             ARM-64 (Native)
Parent Process:        zsh [46560]
Responsible:           Terminal [46558]
User ID:               501

Date/Time:             2024-09-28 13:12:35.7554 +0700
OS Version:            macOS 15.0 (24A335)
Report Version:        12
Anonymous UUID:        59834C9C-83C2-881F-4893-B8F6AD52F92E

Sleep/Wake UUID:       BFC7A2B7-DCAF-485F-940F-5D05919A73D1

Time Awake Since Boot: 36000 seconds
Time Since Wake:       2583 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000001, 0x000000010c035d38

Termination Reason:    Namespace SIGNAL, Code 5 Trace/BPT trap: 5
Terminating Process:   exc handler [46692]

Thread 0 Crashed:: CrBrowserMain Dispatch queue: com.apple.main-thread
0   Electron Framework            	       0x10c035d38 electron::fuses::IsLoadBrowserProcessSpecificV8SnapshotEnabled() + 45844
1   Electron Framework            	       0x10c035d38 electron::fuses::IsLoadBrowserProcessSpecificV8SnapshotEnabled() + 45844
2   Electron Framework            	       0x10c04b6d4 v8::internal::GetCurrentStackPosition() + 35436
3   libswiftCore.dylib            	       0x1927c7a10 _swift_release_dealloc + 56
4   libswiftCore.dylib            	       0x1927c8a64 bool swift::RefCounts<swift::RefCountBitsT<(swift::RefCountInlinedness)1>>::doDecrementSlow<(swift::PerformDeinit)1>(swift::RefCountBitsT<(swift::RefCountInlinedness)1>, unsigned int) + 160
5   libobjc.A.dylib               	       0x18141bd94 AutoreleasePoolPage::releaseUntil(objc_object**) + 204
6   libobjc.A.dylib               	       0x181418138 objc_autoreleasePoolPop + 260
7   Electron Framework            	       0x10f19c3f8 node::GetArrayBufferAllocator(node::IsolateData*) + 17961896
8   Electron Framework            	       0x10f19e738 node::GetArrayBufferAllocator(node::IsolateData*) + 17970920
9   Electron Framework            	       0x10bd84010 0x10bbb8000 + 1884176
10  Electron Framework            	       0x10f19d7c0 node::GetArrayBufferAllocator(node::IsolateData*) + 17966960
11  CoreFoundation                	       0x1818ce414 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 28
12  CoreFoundation                	       0x1818ce3a8 __CFRunLoopDoSource0 + 176
13  CoreFoundation                	       0x1818ce10c __CFRunLoopDoSources0 + 244
14  CoreFoundation                	       0x1818cccf8 __CFRunLoopRun + 840
15  CoreFoundation                	       0x1818cc334 CFRunLoopRunSpecific + 572
16  HIToolbox                     	       0x18cd040cc RunCurrentEventLoopInMode + 292
17  HIToolbox                     	       0x18cd09ebc ReceiveNextEventCommon + 636
18  HIToolbox                     	       0x18cd0a020 _BlockUntilNextEventMatchingListInModeWithFilter + 76
19  AppKit                        	       0x18540fa70 _DPSNextEvent + 660
20  AppKit                        	       0x185d357b8 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688
21  AppKit                        	       0x185402b7c -[NSApplication run] + 480
22  Electron Framework            	       0x10f19f318 node::GetArrayBufferAllocator(node::IsolateData*) + 17973960
23  Electron Framework            	       0x10f19d184 node::GetArrayBufferAllocator(node::IsolateData*) + 17965364
24  Electron Framework            	       0x10f155ed4 node::GetArrayBufferAllocator(node::IsolateData*) + 17673860
25  Electron Framework            	       0x10f11cabc node::GetArrayBufferAllocator(node::IsolateData*) + 17439340
26  Electron Framework            	       0x10e2af164 node::GetArrayBufferAllocator(node::IsolateData*) + 2310420
27  Electron Framework            	       0x10e2b0bfc node::GetArrayBufferAllocator(node::IsolateData*) + 2317228
28  Electron Framework            	       0x10e2acad4 node::GetArrayBufferAllocator(node::IsolateData*) + 2300548
29  Electron Framework            	       0x10c149288 v8::internal::compiler::BasicBlock::set_loop_header(v8::internal::compiler::BasicBlock*) + 13948
30  Electron Framework            	       0x10c14a3f0 v8::internal::compiler::BasicBlock::set_loop_header(v8::internal::compiler::BasicBlock*) + 18404
31  Electron Framework            	       0x10c14a220 v8::internal::compiler::BasicBlock::set_loop_header(v8::internal::compiler::BasicBlock*) + 17940
32  Electron Framework            	       0x10c148acc v8::internal::compiler::BasicBlock::set_loop_header(v8::internal::compiler::BasicBlock*) + 11968
33  Electron Framework            	       0x10c148c6c v8::internal::compiler::BasicBlock::set_loop_header(v8::internal::compiler::BasicBlock*) + 12384
34  Electron Framework            	       0x10be1a2f0 ElectronMain + 128
35  dyld                          	       0x181464274 start + 2840

mvladic, 2024-09-28:
This is already reported in #537 and fixed in the master branch. The new Studio will be released in couple of days.

closed 2024-09-28 by mvladic

LEKPCSTEAM, 2024-09-28:
> This is already reported in #537 and fixed in the master branch. The new Studio will be released in couple of days.

Thank you for your support. 🫡🥰