eez-open/studio #557: eez-flow crash when event handler(flow) on a screen is GESTURE (this i tested)

petrubuciu, 2024-09-28:
**Describe the bug**
i have a ui designed in studio and on every screen i have put event handler: GESTURE, when i touch to enter the screen with gesture event crash event on other screen does not have the event handler setup

**To Reproduce**
Steps to reproduce the behavior:
1. in my main.cpp after the ui_init() and in loop the ui_tick()
2. the ui init perfectly and screen dashboard init and displayed but after i touch to go on other screen like settings(here i put the gesture event hadler my esp32-s3 crash

**Expected behavior**
to work as expected

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [SONOMA]
 - Version [macos 14.7]

** LVGL version**
- 9.2

**Additional context**
Eez Studio 16.1 latest
i have an ESP32-S3 with ILI9488 touch screen 320x480 3.5inch
Here are decoded backtrace:
Stack Trace:
0x4206fc9c: lv_indev_get_gesture_dir at lib/lvgl/src/indev/lv_indev.c line 476
0x4202fb55: flowPropagateValueLVGLEvent(void*, unsigned int, unsigned int, lv_event_t*) (lib/ui/eez-flow 2.cpp:6616).
6616	    int32_t gestureDir = (int32_t)lv_indev_get_gesture_di at )lv_indev_get_gesture_dir( line undefined
0x420396ae: event_handler_cb_dashboard_dashboard (lib/ui/screens.c:23).
23	        flowPropagateValueLVGLEven at flowState, 1, 1, e line undefined
0x42008dc2: lv_event_send (lib/lvgl/src/misc/lv_event.c:98).
98	            dsc->c at e line undefined
0x42019a14: event_send_core (lib/lvgl/src/core/lv_obj_event.c:367).
367	    res = lv_event_sen at list, e, false line undefined
0x42019ac4: lv_obj_send_event (lib/lvgl/src/core/lv_obj_event.c:67).
67	    lv_result_t res = event_send_cor at &e line undefined
0x42005635: lv_screen_load_anim (lib/lvgl/src/display/lv_display.c:724).
724	    if(act_scr) lv_obj_send_even at act_scr, LV_EVENT_SCREEN_UNLOAD_START, NULL line undefined
0x42037617: replacePageHook(int16_t, uint32_t, uint32_t, uint32_t) (lib/ui/eez-flow 2.cpp:6595).
6595	    lv_scr_load_anim(getLvglObjectFromIndex(g_currentScreen), at lv_scr_load_anim_t)animType, speed, delay, false line undefined
0x4202d44b: eez_flow_push_screen(int16_t, lv_screen_load_anim_t, uint32_t, uint32_t) (lib/ui/eez-flow 2.cpp:6550).
6550	    eez::flow::replacePageHoo at screenId, animType, speed, delay line undefined
0x420366b9: eez::flow::executeLVGLComponent(eez::flow::FlowState*, unsigned int) (lib/ui/eez-flow 2.cpp:3787).
3787	                eez_flow_push_screen(specific->screen, at lv_scr_load_anim_t)specific->fadeMode, specific->speed, specific->delay line undefined
0x4202ff19: eez::flow::executeComponent(eez::flow::FlowState*, unsigned int) (lib/ui/eez-flow 2.cpp:3055).
3055				executeComponentFunctio at flowState, componentIndex line undefined
0x42038a78: eez::flow::tick() (lib/ui/eez-flow 2.cpp:6105).
6105	                executeComponen at flowState, componentIndex line undefined
0x42038a9f: eez_flow_tick() (lib/ui/eez-flow 2.cpp:6584).
6584	    eez::flow::tic at  line undefined
0x4202a3ef: ui_tick (lib/ui/ui.c:2038).
2038	    eez_flow_tic at  line undefined
0x42004d44: IPS_ILI9488::init() (src/IPS_ILI9488.cpp:44).
44	        ui_tic at  line undefined
0x42004fd5: setup() (src/main.cpp:31).
31	  display->ini at  line undefined
0x4204661e: loopTask(void*) (/Users/petrubuciu/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42).
42	    setu at  line undefined

petrubuciu, 2024-09-28:
after delete all gesture event handlers, still crash. but after i delete the SCREEN_UNLOAD_START everything worked perfectly

mvladic, 2024-09-28:
Regarding `GESTURE` event check "Backward compatibly notice:" in #510.

Regarding `SCREEN_UNLOAD_START` there is some fixes in the master branch of Studio that could be related to your problem. You can build Studio and try it or you can wait a couple of days when new version of Studio will be released.

Also, you can send here your eez-project file (just zip it and attach in the comment) that cause crashes (with `GESTURE` and `SCREEN_UNLOAD_START` event handlers) so I can try it on the latest studio.

petrubuciu, 2024-09-30:
The problem was not the GESTURE after all(work fine), just SCREEN_UNLOAD_START, but not eez studio crash, the esp32-s3. also, sorry, i can't upload the eez project here. today i will try to rebuild the eez studio from master branch to see if SCREEN_UNLOAD_START resolved and i will report. thx

mvladic, 2024-09-30:
> also, sorry, i can't upload the eez project here

To upload eez-project, just put it in the zip file - githup allows zip files to be uploaded.