eez-open/studio #563: Show zoom level for the active page

zalexzperez, 2024-09-30:
When you start working on multiple pages, it's normal that they end up having different zoom values, especially on small displays.

If you want to visually compare widget positions/sizes between different screens, you need a common zoom value, so having a permanent field showing the current page zoom is of help.

And maybe, a button to reset zoom for all the project pages at once would be a good addition as well.

mvladic, 2024-09-30:
You can double click on empty space to reset zoom and center the screen.

zalexzperez, 2024-09-30:
That's true for the second feature and I actually do that all the time, but I still have 10 "disorganized" screens. I will have to get used to double click before changing to another page. Although I constantly need to zoom in to work on my low-resolution display. Perhaps a general zoom level for all the pages would be better than a global reset zoom button. Like some design programs:
[image]

mvladic, 2024-09-30:
Makes sense. I'm not reading carefully again ;)

mvladic, 2024-10-06:
I'm not sure where to put this zoom level control and how user will choose whether to set the zoom level for the selected page or globally for all pages.

Alternatively, what you want can be done like this:

[image]

What do you think?

zalexzperez, 2024-10-06:
That looks very nice.

Now, I hope you're able to read all of this :P

About the zoom level control, I initially took inspiration in CorelDRAW. This is how it works:

https://github.com/user-attachments/assets/832a8a54-cf96-495c-b0a5-126df2010496

As you can see, both the zoom level and the position are global. 

To be honest, since this feature was about enhancing zoom control to be able to compare widget positions between screens more easily and also about not having a mess of different zoom levels between all the pages, I think it would be best to have a global zoom by default.

By having it like this, user only has to take care of zoom from a single place (page).

About having a global position control, it has its own charm: again, if we want to compare two pages, having the same exact position is helpful. But I can also see an important disadvantage: e.g. you zoom all the way into a remote area of flow actions and go to another page, you will likely end up viewing an empty space. Nothing a quick zoom out can't fix, though.

------------------

So,  I'd do it like this:

1- A current zoom level dropdown with a few predefined values.
2- A button to enable/disable global zoom.

When global zoom is disabled, everything can be as it is now, with the new context menu showing Reset and Set options.
When global zoom button is enabled, all the pages automatically are set to the current's page zoom level. Context menu will still work, but the Set option shouldn't be available for use.

The global position control with an extra button... although nice to have, I think it's a bit overkill.

And finally, about where to place the controls, this would be my preferred location:

[image]

mvladic, 2024-10-07:
I added this dropdown:

[image]

Right click on the empty area when Global Zoom is enabled shows this context menu:

[image]

If Global Zoom is not enabled, it shows:

[image]

"Center view" will only put page in the center, it will not change zoom.
Set the Same Zoom for All Pages" will only affect the zoom, not position.

Doubleclick on the page both center view and sets the zoom to 100%, as before.

zalexzperez, 2024-10-07:
Great job! Love this feature!

It is also worth mentioning that it is possible to quickly center view on all pages as well:

[image]