eez-open/studio #569: Encoder navigation issues in the simulator

zalexzperez, 2024-10-02:
To make it clearer for future readers: Encoder = mouse wheel. Encoder button: middle mouse-wheel button.

Found problems:

1) Slider widget in edit mode: if value was not changed after entering the edit mode, it's not possible to return to navigate mode by pressing on the encoder button. 

2) Once right out of edit mode, if you press on the encoder button to return to edit mode, it's not possible unless we refocus on the slider by focusing on another object first.

3) Suppose you are focusing on a button and encoder button is hold-pressed, `LV_INDEV_STATE_PRESSED` is presumably set, the button gets in the `FOCUS-KEY` state. Once released, nothing happens; button is still pressed.
But, if we rotate the encoder to focus on another group object, then that seems to finally set a `LV_INDEV_STATE_RELEASED` state because the button is unpressed and the action is performed.

In conclusion, all three issues are fixed/prevented by rotating the encoder.

mvladic, 2024-10-06:
Fixed.

zalexzperez, 2024-10-07:
Encoder navigation inside dropdown requires

closed 2024-10-07 by zalexzperez