eez-open/studio #571: Failed build on Arduino IDE (GIGA Display)

andrearizzello, 2024-10-09:
**Describe the bug**
When i try to compile a simple project on  the arduino IDE i get this error in the title 

**To Reproduce**
- Create a new blank arduino project
- Create an interface inside an EEZ Studio project (LVGL with EEZ flow activated) on EEZ Studio
- Build on EEZ Studio and put all the build files inside the arduino library folder (~Arduino/libraries/ui/ALL_THE_FILES_ARE_HERE)
- Compile on  Arduino IDE
- Fail

**Expected behavior**
A successful compilation

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Macos
 - Version: Ventura 13.7

**LVGL version (if used)**
- 9.1.0

**Additional context**
The problem  doesn't present when i turn off EEZ Flow

**Full compilation logs**
[LOGS](https://pastebin.com/fBKE5qxq)

andrearizzello, 2024-10-09:
Hello @mvladic i get the same error without :/ :
```
In file included from /Users/andrea/Documents/Arduino/libraries/ui/ui.h:6:0,
                 from /Users/andrea/Downloads/sketch_testttttt/sketch_testttttt.ino:6:
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.h:1880:1: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
 const uint32_t getThemeColorsCount(int themeIndex);
 ^~~~~
In file included from /Users/andrea/Documents/Arduino/libraries/ui/eez-flow.cpp:11:0:
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.h:1880:1: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
 const uint32_t getThemeColorsCount(int themeIndex);
 ^~~~~
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.cpp: In function 'bool eez::decompressAssetsData(const uint8_t*, uint32_t, eez::Assets*, uint32_t, int*)':
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.cpp:295:6: warning: unused variable 'compressedSize' [-Wunused-variable]
  int compressedSize = assetsDataSize - compressedDataOffset;
      ^~~~~~~~~~~~~~
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.cpp: At global scope:
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.cpp:440:1: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
 const uint32_t getThemeColorsCount(int themeIndex) {
 ^~~~~
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.cpp: In function 'void eez::flow::executeLabelOutComponent(eez::flow::FlowState*, unsigned int)':
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.cpp:3514:47: warning: comparison is always true due to limited range of data type [-Wtype-limits]
     if ((int)component->labelInComponentIndex != -1) {
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.cpp: In function 'void eez::flow::objGetX(eez::flow::FlowState*, unsigned int, const eez::ListOfAssetsPtr<eez::Property>&, uint32_t)':
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.cpp:4266:45: error: call of overloaded 'Value(int32_t&, eez::ValueType)' is ambiguous
     RESULT(result, Value(x, VALUE_TYPE_INT32));
                                             ^
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.cpp:4235:57: note: in definition of macro 'RESULT'
     assignValue(flowState, componentIndex, NAME##Value, VALUE);
                                                         ^~~~~
In file included from /Users/andrea/Documents/Arduino/libraries/ui/eez-flow.cpp:11:0:
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.h:991:2: note: candidate: eez::Value::Value(double, eez::ValueType)
  Value(double value, ValueType type)
  ^~~~~
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.h:987:5: note: candidate: eez::Value::Value(float, eez::ValueType)
     Value(float value, ValueType type)
     ^~~~~
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.h:975:5: note: candidate: eez::Value::Value(uint64_t, eez::ValueType)
     Value(uint64_t value, ValueType type)
     ^~~~~
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.h:971:5: note: candidate: eez::Value::Value(int64_t, eez::ValueType)
     Value(int64_t value, ValueType type)
     ^~~~~
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.h:967:5: note: candidate: eez::Value::Value(uint32_t, eez::ValueType)
     Value(uint32_t value, ValueType type)
     ^~~~~
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.h:963:5: note: candidate: eez::Value::Value(uint16_t, eez::ValueType)
     Value(uint16_t value, ValueType type)
     ^~~~~
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.h:959:5: note: candidate: eez::Value::Value(int16_t, eez::ValueType)
     Value(int16_t value, ValueType type)
     ^~~~~
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.h:955:5: note: candidate: eez::Value::Value(uint8_t, eez::ValueType)
     Value(uint8_t value, ValueType type)
     ^~~~~
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.h:951:5: note: candidate: eez::Value::Value(int8_t, eez::ValueType)
     Value(int8_t value, ValueType type)
     ^~~~~
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.h:943:5: note: candidate: eez::Value::Value(int, eez::ValueType)
     Value(int value, ValueType type)
     ^~~~~
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.cpp: In function 'void eez::flow::objGetY(eez::flow::FlowState*, unsigned int, const eez::ListOfAssetsPtr<eez::Property>&, uint32_t)':
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.cpp:4284:45: error: call of overloaded 'Value(int32_t&, eez::ValueType)' is ambiguous
     RESULT(result, Value(y, VALUE_TYPE_INT32));
                                             ^
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.cpp:4235:57: note: in definition of macro 'RESULT'
     assignValue(flowState, componentIndex, NAME##Value, VALUE);
                                                         ^~~~~
In file included from /Users/andrea/Documents/Arduino/libraries/ui/eez-flow.cpp:11:0:
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.h:991:2: note: candidate: eez::Value::Value(double, eez::ValueType)
[... 154 lines cut ...]
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.cpp: In function 'void eez::flow::setGlobalVariable(eez::Assets*, uint32_t, const eez::Value&)':
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.cpp:6922:29: warning: comparison of unsigned expression >= 0 is always true [-Wtype-limits]
[... cut ...]
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.cpp:7428:54: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     int32_t userData = *((int32_t*)(&userDataPointer));
                                                      ^
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.cpp: At global scope:
/Users/andrea/Documents/Arduino/libraries/ui/eez-flow.cpp:7278:16: warning: 'int32_t getLvglScreenByName(const char*)' defined but not used [-Wunused-function]
 static int32_t getLvglScreenByName(const char *name) {
                ^~~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1
```

mvladic, 2024-10-09:
The same error was reported today on Discord also:

[image]

This is now fixed. A new HF release 0.18.1 will be released to day.

andrearizzello, 2024-10-09:
That's awesome, thanks! 🙏🏼

closed 2024-10-09 by mvladic