eez-open/studio #580: EEZ crashes (MacOS) within 10-15 seconds of loading Roboto Regular.ttf font in size 172 for numbers

yaredna, 2024-10-10:
**Describe the bug**
Created a LVGL project, one screen, one label with the text "00.0". Loaded a google font Roboto Regular.ttf, only numbers (32, 42-58) in size 172. The EEZ editor crashes with a blank unresponsive screen within 10-15 seconds from creating the font. 

**To Reproduce**
Create new LVGL 8.3 project. Create a Roboto_172 font using Google's Roboto Regular, size 172, range (32, 42-58). Change the text of the Hello, world! label to "00.0", and font to Roboto_172. Keep working. The editor will crash 100% of the time within 15 seconds with a white screen, unresponsive. 

**Expected behavior**
Expec the program to continue working as the font loaded properly. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Mac OS
 - Version : Sonomo 14.6.1

** LVGL version (if used)** 

- 8.3

mvladic, 2024-10-10:
It doesn't seem to crash on Windows, I'll test on Mac tomorrow.

[image]

yaredna, 2024-10-11:
> It doesn't crash on my Mac too (however, I have macOS 15).
>
> Happy to send you the files if you need.
>
> Yes, please, send me your eez-project file.

Let try the same example on a different version but I will also try it on my wife’s laptop too. Thanks so much !! Will send the files if I can repeat the crash on my wife’s Mac !!Sent from my iPhone

yaredna, 2024-10-11:
Excellent. After receiving your message, I went ahead and test it on my wife’s Intel Mac, and it worked. I will try figuring out why it crashes on my system: must be related to my installation. 

Let me know please;
1. Do you want me to delete the post or should I update it there is no issue 
2. The project that crashed my system is attached. Let me know if you need other files. 

Thanks so much !!!!!!
Nadim
￼

mvladic, 2024-10-11:
I don't see attached project. You should zip your eez-project file before attaching here on github.

yaredna, 2024-10-12:
Sorry ! Here is the zip file attached. Do you also need the files generated by the export too ? 
￼

yaredna, 2024-10-12:
[WindAlarm.eez-project-ui-state.zip](https://github.com/user-attachments/files/17348414/WindAlarm.eez-project-ui-state.zip)

The email interface to this forum doesn't permit attachments. Sorry. I am attaching the file hereby. I will edit the original post to reflect that it worked fine on another Mac, and could be configuration specific.

yaredna, 2024-10-12:
I believe now that the issue is specific to my personal Mac, probably a configuration issue. It is working fine on my wife's Mac. I included the zip file here just in case, but I am closing this issue. Great response from the team, super impressed !! Moving my entire project from Squareline to EEZ is likely to happen in the new few days.

closed 2024-10-12 by yaredna