eez-open/studio #582: Action widget distinction for native user actions

zalexzperez, 2024-10-12:
Flow and Native user actions share the same looks.

To have more information, I propose a distinguished look for native actions, for example:

[image]

mvladic, 2024-10-12:
Implemented.

zalexzperez, 2024-10-12:
I appreciate the change. I don't want to be annoying, but please, consider this.

Right now, the User Actions section looks like this:

[image]

But you may not remember what the lighter or darker green stands for. If you hover over the widget, you also don't get information about whether it's a native or a flow action.

This gets worse when you have the widget in the design area, without anything to compare it with:

[image]

To be able to easily tell what type of action it is at first glance, I'd add an extra touch to help with that, without confusing anyone. Two ideas:

- Something inside the parenthesis: `(/)` or `(#)`
- Something more subtle: for example, using curly braces for native actions: 
[image]

I particularly prefer the last one. If you don't like any of the two, that's fine.

Also, when selecting the user action, properties panel could show it's a native action:

[image]

And finally, like mentioned before, hovering mouse cursor over the action, could indicate it as well:
[image]

mvladic, 2024-10-31:
> Something more subtle: for example, using curly braces for native actions

Implemented.

closed 2024-11-04 by mvladic