eez-open/studio #591: strange behavior in arrays inside custom struct

mircotaddei, 2024-10-15:
hello, i am using eezstudio for my hobbyist work.
eezstudio is well done, many thanks for this great project.

OS: Linux / ESP32 - ili9341
ESP32 / arduino framework
LVGL:l 9.2

I defined a custom  struct "Program" with "days" as field of vector of boolean.
In a new screen I created a switch directly connected to a vector element.

The problem I encountered is that once selected the value always remains true even if the switch state is off.

The last three screenshots show the state of the vector in the initial state, after the first click and after the second.

[image]
[image]
![Screenshot_2024-10-15_07-59-25](https://github.com/user-attachments/assets/5ad8dd9e-ab1c-4352-a6b8-52291b945
[image]
451)
[image]

mircotaddei, 2024-10-15:
[image]

the first log screenshot was not loaded...

mvladic, 2024-10-15:
I can't reproduce this problem. Maybe I'm doing something differently. Here is my project:

[test_switch_with_array_boolean.zip](https://github.com/user-attachments/files/17382213/test_switch_with_array_boolean.zip)

I created a Program structure similar to yours:

[image]

I created a global variable `programs`:

[image]

Here is a Switch widget properties:

[image]

And here is the result:

[image]

mircotaddei, 2024-10-16:
the problem is not in eez-studio but I think in the workflow when I try to read array values ​​from my C language sources.

here is my callback that read values from flow and try to save in my classes:

`
// ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++ PROGRAMS SAVE ++++++++++
void action_programs_save(lv_event_t *e)
{
    LOG_DEBUG("action_programs_save START");

    ArrayOfProgramValue progs = flow::getGlobalVariable(FLOW_GLOBAL_VARIABLE_PROGRAMS);
    for (int i = 0; i < progs.size(); i++)
    {
        ProgramValue prog = progs.at(i);
        LOG_TRACE("i=%d, name=%s, enabled=%d, status=%d, hours=%d, mins=%d", i, prog.name(), prog.enabled(), prog.status(), prog.hours(), prog.mins());
        for (int j = 0; j < prog.days().size(); j++)
        {
            LOG_TRACE("days[%d]=%d", j, prog.days().at(j));
        }
        ArrayOfBoolean p=  prog.zones_enabled();
        LOG_TRACE("zones_enabled=%p", &p);
        for (int j = 0; j < prog.zones_enabled().size(); j++)
        {
            LOG_TRACE("zones_enabled[%d]=%d", j, prog.zones_enabled().at(j));
            LOG_TRACE("p[%d]=%d", j, p.at(j));
        }
        // LOG_TRACE("zones_duration=%p", &prog.zones_duration);
        for (int j = 0; j < prog.zones_duration().size(); j++)
        {
            LOG_TRACE("zones_duration[%d]=%d", j, prog.zones_duration().at(j));
        }
    }

    for (int i = 0; i < SPR_MAX_PROGRAMS; i++)
    {
        LOG_TRACE("action_programs_save: i=%d, name=%s, enabled=%d, status='%d'", i, progs.at(i).name(), progs.at(i).enabled(), progs.at(i).status());
        bool zones_enabled[SPR_MAX_ZONES];
        int zones_duration[SPR_MAX_ZONES];
        for (int j = 0; j < SPR_MAX_ZONES; j++)
        {
            LOG_TRACE("zones_enabled[%d]=%d, zones_duration[%d]=%d", j, progs.at(i).zones_enabled().at(j), j, progs.at(i).zones_duration().at(j));
            zones_enabled[j] = progs.at(i).zones_enabled().at(j);
            zones_duration[j] = progs.at(i).zones_duration().at(j);
        }
        int hours = progs.at(i).hours();
        int minutes = progs.at(i).mins();

        bool days[7];
        for (int j = 0; j < 7; j++)
        {
            days[j] = progs.at(i).days().at(j);
            LOG_TRACE("days[%d]=%d", j, days[j]);
        }
        bdays_t bdays = boolArrayToBdays(days);
        LOG_TRACE("bdays=%d", bdays);
        Sprinkler.setProgram(i,
                             progs.at(i).name(),
                             progs.at(i).enabled(),
                             zones_enabled,
                             zones_duration,
                             (byte)progs.at(i).hours(),
                             (byte)progs.at(i).mins(),
                             bdays);
        // TODO cron
    }

    LOG_DEBUG("action_programs_save END");
}
`

mvladic, 2024-10-16:
Thanks! I will check this.

mircotaddei, 2024-10-17:
Thank you so much!!