eez-open/studio #597: Two little problems about gesture event

liriyao, 2024-10-18:
1. gesture event is too easy to trigger .
    Less than 20 pixels will trigger gesture event and i don't find any place to change the length myself
2. When first click on an obj and try to trigger gesture event , the obj press event will trigger too.
   I wanted to do the same thing as a gesture swipe on a phone. It is anyone slove the same problem or any suggestions. Thanks

mvladic, 2024-10-18:
In `lv_indev_t * indev` there is `gesture_limit` and `gesture_min_velocity` fileds. They are initialized with these values:

```
/*Gesture threshold in pixels*/
#define LV_INDEV_DEF_GESTURE_LIMIT        50

/*Gesture min velocity at release before swipe (pixels)*/
#define LV_INDEV_DEF_GESTURE_MIN_VELOCITY 3
```

You can't change these fields if you include `<lvgl/lvgl.h>`, but you can include `<lvgl/lvgl_private.h>`.

mvladic, 2024-10-18:
> When first click on an obj and try to trigger gesture event , the obj press event will trigger too.
I wanted to do the same thing as a gesture swipe on a phone. It is anyone slove the same problem or any suggestions. Thanks

Do not use PRESSED event of the button, use CLICKED.

liriyao, 2024-10-18:
> > When first click on an obj and try to trigger gesture event , the obj press event will trigger too.
> > I wanted to do the same thing as a gesture swipe on a phone. It is anyone slove the same problem or any suggestions. Thanks
> 
> Do not use PRESSED event of the button, use CLICKED.

CLICKED don't help . i will trigger too

> In `lv_indev_t * indev` there is `gesture_limit` and `gesture_min_velocity` fileds. They are initialized with these values:
>
> ```
> /*Gesture threshold in pixels*/
> #define [...]
 
i will try this one . thanks for your reply

mvladic, 2024-10-18:
> CLICKED don't help . i will trigger too

I found this LVGL issue: https://github.com/lvgl/lvgl/issues/3211. It is suggested to call `lv_indev_wait_release(lv_indev_active());` in gesture event handler to prevent `CLICKED`. 

If you are using LVGL project without EEZ Flow you can just make this call in your event handler for the gesture event. 

If you are using EEZ Flow then currently there is no way to do this. We can add call to `lv_indev_wait_release` for the GESTURE event always but I'm not sure this is always desirable.

mvladic, 2024-10-18:
> If you are using EEZ Flow then currently there is no way to do this. We can add call to lv_indev_wait_release for the GESTURE event always but I'm not sure this is always desirable.

I just tested this. I modified EEZ Flow engine to call `lv_indev_wait_release(lv_indev_active())` on GESTURE event and here is the result:

[image]

mvladic, 2024-10-18:
Maybe we should indeed add this in the EEZ Flow engine. Way would anyone want CLICKED event when GESTURE is also detected.

mvladic, 2024-10-18:
I added `lv_indev_wait_release(lv_indev_active());` in the GESTURE event handler when EEZ Flow is used. This is now committed in the master branch of Studio.

liriyao, 2024-10-18:
> I added `lv_indev_wait_release(lv_indev_active());` in the GESTURE event handler when EEZ Flow is used. This is now committed in the master branch of Studio.

I will  update my code and try. it help a lot when the screen got a lot of obj,but need to do slide to change screen. thanks

closed 2024-11-04 by mvladic