eez-open/studio #598: Could you add a code editing feature and enable code suggestions/intelliSense using VS Code?

JJZBQA, 2024-10-18:
Could you add a code editing feature and enable code suggestions/intelliSense using VS Code?

JJZBQA, 2024-10-18:
Is it possible to open an LVGL architecture that doesn't use eez_flow, allowing the creation of animations with variables and more?

zalexzperez, 2024-10-18:
I think adding intelliSense would slow down the program. It would also need to have access to all your project code...I don't see.

What I would like to see, though, is the following:
when working on a file from VScode, say, vars.cpp, once overwritten, Studio could detect this change and pick up the changes, placing them into the editor.

mvladic, 2024-10-18:
> Could you add a code editing feature and enable code suggestions/intelliSense using VS Code?
> Is it possible to open an LVGL architecture that doesn't use eez_flow, allowing the creation of animations with variables and more?

Adding a full C/C++ code editing feature/experience is currently out of scope. The idea is that you Build studio generated code and continue work with your C/C++ editor/IDE of choice.

> What I would like to see, though, is the following:
when working on a file from VScode, say, vars.cpp, once overwritten, Studio could detect this change and pick up the changes, placing them into the editor.

Yes, if `vars.cpp` is in studio project file templates list we could do that.

JJZBQA, 2024-10-18:
Thank you, that would be great!