eez-open/studio #605: TextArea, Floating point var, Keypad LVGL 8.0 and 9.0

Sanjaypatel145, 2024-10-19:
**Describe the bug**
When editing the text area with Numeric Key board decimal point can be placed at last while editing, 
but when text area is assigned to variable using expression decimal point cannot be placed at the last place.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

Working correctly when text area is literal
[image]

error here when text area is assigned to variable data using expression 
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 10

** LVGL version (if used)**

- 8.x or 9.x both

**Additional context**
Add any other context about the problem here.

mvladic, 2024-10-31:
In Textarea widget, "Value" property should be bound to string variable and you can use `Flow.parseFloat` to convert string to float.

[lvgl_debug_keyboard_textarea_decimal_point.zip](https://github.com/user-attachments/files/17585192/lvgl_debug_keyboard_textarea_decimal_point.zip)

[image]

Sanjaypatel145, 2024-11-06:
fixed

closed 2024-11-06 by Sanjaypatel145

Sanjaypatel145, 2024-11-15:
Still a problem occurs when user accidentally inserts "." two times the flow will throw error, eventually rebooting....

can you please suggest me to resolve this

mvladic, 2024-11-15:
You can catch that error. Here is the updated project that does that:

[lvgl_debug_keyboard_textarea_decimal_point.zip](https://github.com/user-attachments/files/17777709/lvgl_debug_keyboard_textarea_decimal_point.zip)

Sanjaypatel145, 2024-11-15:
> BTW to catch an error in some action enable "Catch error" option:
>
> image.png (view on web)

Thanks a lot, let me try and will update you on the same.