eez-open/studio #606: ImgButton, This widget type is not supported in LVLG 9.0

LonlyPan, 2024-10-20:
**Describe the bug**
ImgButton,  check 。log：This widget type is not supported in LVLG 9.0
But it is ok when Run and 
is ok too when work on my esp32 c3 board 

**To Reproduce**
Steps to reproduce the behavior:
1. Add a ImgButton to the project
2. checks
3. error,like this
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version ：0.18.2

** LVGL version (if used)**
-  9.x

mvladic, 2024-10-31:
Fixed.

flixmyswitch, 2024-10-31:
Nice. When do you think we will see V0.19.0

mvladic, 2024-10-31:
Version 0.19.0 should be released next week. 

BTW some people build Studio from source. [Here are the instructions](https://github.com/eez-open/studio?tab=readme-ov-file#build-and-run-from-source-all-operating-systems).