eez-open/studio #613: [LVGL] Tables

ajayre, 2024-11-02:
LVGL 9.1

How can I populate and style tables in the editor? Is there a place to enter custom LVGL API code and have it execute with the Run button?

mvladic, 2024-11-02:
Not possible right now from the Studio. In Studio, you can only create Table widget, the rest must be done from the code.

closed 2024-11-02 by mvladic

ajayre, 2024-11-02:
I edited ui.c and added to the end of ui_init() the following:

```
    lv_table_set_row_count(TimingTable, 3);
    lv_table_set_column_count(TimingTable, 3);
    lv_table_set_cell_value(TimingTable, 1, 1, "Content");
```

It checks and builds OK but when I Run it in EEZ Studio these lines have no effect. Am I adding them in the wrong place? Or is simulating the table in EEZ Studio not possible at all?

Thanks!

mvladic, 2024-11-02:
What is `TimingTable`, how this variable is initialized?

ajayre, 2024-11-02:
It is the name of the table widget.

I now tried this:

Created a native user action called ScreenLoaded
Went to the screen properties
Added a SCREEN_LOADED event handler, type action, action ScreenLoaded
Created action.cpp with the following contents:

```
#include <lvgl/lvgl.h>

#include "actions.h"
#include "screens.h"
#include "ui.h"

void action_ScreenLoaded(lv_event_t *e)
{
    lv_table_set_row_count(TimingTable, 3);
    lv_table_set_column_count(TimingTable, 3);
    lv_table_set_cell_value(TimingTable, 1, 1, "Content");
}
```

Checks and builds OK but doesn't seem to affect the table. I must be missing a step?

Thanks.

mvladic, 2024-11-02:
Usually, widget objects are accessed with `objects.my_widget_name` where my_widget_name is entered in `Name` property of the widget.

ajayre, 2024-11-02:
Thanks! I changed it to:

`objects.injector_timing_table`

and it still doesn't work. I then added some junk to the action.cpp file and it still checks, builds and runs without error. So it seems this file is not being used at all?

I then added some junk to ui_init() and same thing - checks, builds and runs. I am confused.

`void ui_init() {
    create_screens();
    loadScreen(SCREEN_ID_MAIN);
    fffff
}
`

mvladic, 2024-11-02:
If you are using CMake then maybe you should run cmake command to include newly added source files.

ajayre, 2024-11-02:
I am just clicking on the Build button in EEZ Studio. Does that not compile the files each time?

mvladic, 2024-11-02:
Studio build just builds/generates source files to be included in your C/C++ project. You need to make sure these source file are include in your Makefile, but this depends how exactly you are building your C/C++ project.

ajayre, 2024-11-02:
So right now I don't have a target. I am only creating the UI and simulating it in EEZ Studio. So it seems I can't do anything with the table widget because EEZ Studio doesn't use the source files at all, right?

mvladic, 2024-11-02:
Yes. These changes in actions.cpp you can only test on your device or you can build your project using Emscripten and test it inside web browser - check this repository how to do that https://github.com/lvgl/lv_web_emscripten.

ajayre, 2024-11-02:
Oh, ok, thanks, Would be nice if full graphical support of creating tables was included because my application really needs them.