eez-open/studio #614: [LVGL] Triggering drop down list event when changing the selected value from a flow node

ajayre, 2024-11-02:
LVGL 9.1

I have a drop down list called Foo
The Selected field of Foo is set to assignable with the global native variable name FooValue
Foo has a VALUE_CHANGED flow event that enables or disables some UI elements depending on if FooValue is 0 or 1
On another page I have a switch.
If the switch is checked then it sets FooValue to 0

When I run this and turn the switch on the drop down list Foo changes to item 0 as expected. However that doesn't seem to trigger the VALUE_CHANGED event which I need to keep the UI consistent.

How do I get the event working?

Thanks!

ajayre, 2024-11-02:
OK, I solved this by adding a SCREEN_LOADED event and in that event I re-evalulate FooValue to force an update of the enabled/disabled items.

But shouldn't setting a variable connected to the Selected property of a drop down list always trigger the VALUE_CHANGED event?

mvladic, 2024-11-02:
This depends on LVGL because LVGL is one who sends VALUE_CHANGED events. When variable is changed with SetVariable, Studio generated code will call `lv_dropdown_set_selected` and this function doesn't send VALUE_CHANGED event.

Hint: You can use Watch action to detect that some variable has been changed and do the stuff.

closed 2024-11-04 by mvladic