eez-open/studio #617: [LVGL] Centering label

ajayre, 2024-11-02:
LVGL 8.x

I have a text label whose value is an expression and it shows "{VersionDisplay}". The label is centered so that this text is centered.

The content comes from a native string variable which has the value "Version 1.00". Because this is shorter than "{VersionDisplay}" the result is slightly off center.

How can I change the text in a label from a string variable and keep it centered?

I tried increasing the width of the label to 480px so it spans the entire width of the screen and then change the align property on the label to CENTER or TOP_MID but it does unexpected things like moving the entire label off screen.

Thanks.

mvladic, 2024-11-02:
> I tried increasing the width of the label to 480px so it spans the entire width of the screen and then change the align property on the label to CENTER or TOP_MID but it does unexpected things like moving the entire label off screen.

This is one way how to center it. I don't know why it does unexpected things without looking at your eez-project file.

Here is another way how Label can be centered by using Align=CENTER style property:

[lvgl_center_label.zip](https://github.com/user-attachments/files/17608150/lvgl_center_label.zip)

ajayre, 2024-11-02:
OK, thanks! I see now. This is what happens:

- Create label
- Set label value to {VersionDisplay}
- Center label on screen
- Set label text align to CENTER

At this point the label jumps off the screen to the bottom, right.

- Drag label back to center of screen

Now it works.

mvladic, 2024-11-02:
Yes, beside changing align to CENTER you need also to set X and Y to zero.

closed 2024-11-04 by mvladic