eez-open/studio #618: Manipulation of flow node view with touchpad

ajayre, 2024-11-02:
This is what I currently am working with:

[image]

It would be nice if there were scrollbars for the view area so I can zoom in and then pan around. Also a zoom to fit everything would be nice.

If I right-click and choose Center View then it wastes a lot of space on the left.

Perhaps this is easier with a mouse but I am using a laptop with a touchpad and it is getting a bit unweildy and painful to use.

Thanks.

mvladic, 2024-11-03:
> Perhaps this is easier with a mouse but I am using a laptop with a touchpad and it is getting a bit unwieldy and painful to use.

With the mouse you can use middle or right button to pan around. With the touchpad I only tested with the Mac and panning with two fingers and pinch to zoom works good in my opinion. I just tested on the Linux Acer laptop and panning with two fingers works, but as you say it is unwieldy to use, i.e. it is to sensitive. On this laptop pinch to zoom doesn't work at all.

mvladic, 2024-11-03:
Panning and zooming on Lenovo Thinkpad laptop on Windows also works good. What operating system are you using?

ajayre, 2024-11-03:
Windows 11.

Pinch to zoom works. Two fingers to pan doesn't work.

mvladic, 2024-11-03:
> Two fingers to pan doesn't work.

Does it not work at all or does it sometimes mix panning and zooming (because both gestures use two fingers)?

ajayre, 2024-11-03:
Doesn't do anything at all.

mvladic, 2024-11-03:
Hm, it works on my Lenovo Thinkpad X1 Extreme WIndows 11 laptop. Does two finger panning/scrolling work in other programs?

mvladic, 2024-11-03:
>Hm, it works on my Lenovo Thinkpad X1 Extreme WIndows 11 laptop

On this laptop I have an option to enable/disable two finger scrolling:

[image]

ajayre, 2024-11-03:
Yes. I couldn't live without it. For example works in Chrome, Notepad++, Visual Studio, etc.

ajayre, 2024-11-03:
Note that two-finger panning would be nice but I am perfectly happy with scroll bars.

mvladic, 2024-11-03:
Ok, I will investigate how scroll bars can be added. There is a complications because page view editor can grow indefinitely in all directions.

I'm curious, what is your laptop model? Maybe I can find a similar laptop to reproduce this problem.

mvladic, 2024-11-03:
BTW on my Thinkpad laptop, panning also works with one finger if I keep pressed right touchpad button (right button press is detected when I press touchpad on the bottom right corner of the touchpad).

ajayre, 2024-11-03:
Yes, I understand the technical difficulty...

I just tried it, right-click and drag does work for panning.

ajayre, 2024-11-03:
Instead of scrollbars how about toolbar buttons for zoom and pan modes?

mvladic, 2024-11-04:
For now, I made a change so you can pan around with the ALT key + "single finger drag".

closed 2024-12-16 by mvladic