eez-open/studio #620: Include font fallback

JomaImport, 2024-11-03:
In many cases it is very helpful to have a fallback for every font, I guess it will be a good idea and very easy to implement, just give the user the possibility to select the fallback font.

mvladic, 2024-11-03:
Are we talking about Dashboard project or LVGL project?

JomaImport, 2024-11-03:
LVGL 8.X

    .dsc = &font_dsc,          /*The custom font data. Will be accessed by `get_glyph_bitmap/dsc` */
    .fallback = NULL,
    .user_data = NULL

In the generated code in the font C file the .fallback is always NULL, it would be helpful if the user could select the fallback font at design time

mvladic, 2024-11-03:
I didn't know about that. Do you know under what circumstances this fallback font will be used?

JomaImport, 2024-11-03:
A fallback is used when one character is not found in the current font, in that case is used the character in the fallback font, so can be used in many cases when the needed character is not found in the generated font.

As an example you can see the two next pictures, the first one is a messagebox using the deult Montserrat font, the second picture uses a generated font and as you will see the X symbol for closing it is not in the generated font so is not displayed. If the Montserrat font is define as fallback font then this symbol is displayed correctly.

For your information we are using ESP-IDF for ESP32-S3.

[image]
[image]

JomaImport, 2024-11-03:
The same issue happens with other widgets like dropdown 

[image]
[image]

mvladic, 2024-11-03:
All clear, thanks!

mvladic, 2024-11-09:
Implemented.

[image]

JomaImport, 2024-11-10:
Thanks for implementing so quickly!

closed 2024-11-18 by mvladic