eez-open/studio #621: Add name to scales and indicators

JomaImport, 2024-11-03:
I don't see the way to access to scales and indicators for meters on the generated code because I can't define a name at design time to them and they are not stored in the objects collection on the generated code, so I can't access to them because the pointer to them is always "scale" and "indicator" and these pointers are overwritten when next scales and indicators are created.

It would be very helpful and I guess easy to implement the name for scales and indicators at design time and add them to the objects collections on the generated code.

If I'm wrong and there is a easy way to access to these scales and pointers then it would be helpful if someone can explain me how to do it.

JomaImport, 2024-11-03:
Thanks for your quick answer, it will be really very helpful

mvladic, 2025-02-10:
The reason this doesn't have a higher priority is that the Meter Widget no longer exists in LVGL v9.x. It is replaced with the Scale widget and there are some issues with the Scale widget that we still need to address.

As a workaround, you can get the scale and indicator objects in the following way:

```C
#include "actions.h"
#include "screens.h"

void action_test(lv_event_t *e) {
    // "watch" meter object is from "Meter" example, it has one scale and 3 indicators:
    lv_meter_t *meter = (lv_meter_t *)objects.watch;
    
    // get scale
    lv_meter_scale_t *scale1 = _lv_ll_get_head(&meter->scale_ll);
    LV_LOG("scale1: %p\n", scale1);
    
    // get indicators
    lv_meter_indicator_t *indic1 = _lv_ll_get_head(&meter->indicator_ll);
    LV_LOG("indic1: %p, %p\n", indic1, indic1->scale);

    lv_meter_indicator_t *indic2 = _lv_ll_get_next(&meter->indicator_ll, indic1);
    LV_LOG("indic2: %p, %p\n", indic2, indic2->scale);

    lv_meter_indicator_t *indic3 = _lv_ll_get_next(&meter->indicator_ll, indic2);
    LV_LOG("indic3: %p, %p\n", indic3, indic3->scale);

    // Additionally, here is an example how to iterate over scales and indicators:
    lv_meter_scale_t *scale_iter;
    _LV_LL_READ_BACK(&meter->scale_ll, scale_iter) {
        // interate over indicators
        lv_meter_indicator_t *indic_iter;
        _LV_LL_READ_BACK(&meter->indicator_ll, indic_iter) {
            // indic is of type "lv_meter_indicator_t *",
            // and indic->scale is parent scale
            
            if (indic_iter->scale == scale_iter) {
                // ...
            }
        }
    }
}
```

prasimix, 2025-02-10:
> I'm a bit dissapointed that this feature is being delayed. We use many meters in our projects, so if we can't access the indicators then it's quite useless. Could you confirm an estimated date to implement this feature. In any case an awesome software!

Thanks for the compliment. Please take into account that EEZ Studio is still financed by donations (individual or funds like NLnet), and all publicly reported tasks are solved according to the best effort principle. Therefore, we only had estimates for tasks that were part of a Milestone agreed with some of the donors or clients who use our professional service.

JomaImport, 2025-02-10:
Thanks for your answer!

JomaImport, 2025-02-12:
I tried your workaround and it works like a charm!
Thanks for your great support, I will start making donations to support the project because it's really good.

mvladic, 2026-02-02:
It is now possible to give the name to scales and indicators in LVGL Meter widget:

[image]

And the following code will be generated in `screens.h`:

```
typedef struct {
    lv_meter_scale_t *pie_scale;
    lv_meter_indicator_t *pie_indicator_1;
    lv_meter_indicator_t *pie_indicator_2;
    lv_meter_indicator_t *pie_indicator_3;
    lv_meter_scale_t *watch_scale;
    lv_meter_indicator_t *hour_indicator;
    lv_meter_indicator_t *min_indicator;
    lv_meter_indicator_t *sec_indicator;
    lv_meter_scale_t *arcs_scale;
    lv_meter_indicator_t *arc_indicator_1;
    lv_meter_indicator_t *arc_indicator_2;
    lv_meter_indicator_t *arc_indicator_3;
    lv_meter_scale_t *needle_scale;
    lv_meter_indicator_t *needle_indicator_1;
    lv_meter_indicator_t *needle_indicator_2;
    lv_meter_indicator_t *needle_indicator_3;
    lv_meter_indicator_t *needle_indicator_4;
    lv_meter_indicator_t *needle_indicator_5;
} screen_main_state_t;

extern screen_main_state_t screen_main_state;
```

So, you can access it with:

```
screen_main_state.watch_scale;
screen_main_state.hour_indicator;
// ...
```

JomaImport, 2026-02-02:
Good news!
I’m really looking forward to being able to try this feature when it’s released.

JomaImport, 2026-02-07:
I have tested this new function, it's working as expected on projects withou themes, but on projects using themes it shows a compilation error in screens.c. I attach screenshot.

[image]

JomaImport, 2026-02-08:
Tested and working 100%
Great job and great software!

I would like to make a donation to show my support, please confirm link to make donations.