eez-open/studio #629: [LVGL] Groups index problem

elmonic, 2024-11-12:
hi thank for new version
eez studio When creating a group, the first widget is wrongly placed in the last widget.
[image]

mvladic, 2024-11-12:
I don't know how to reproduce this problem. Can you send me the eez-project file in which this problem occurs?

elmonic, 2024-11-12:
For example, you create a group and then create several series of widgets. Now add a group to the widgets and specify the number of each widget in the group.
After this, you will see that widget number zero is placed in the group list as the last widget. Now, when the code is created, the first widget is placed at the end.

mvladic, 2024-11-12:
So, if you want to use "Group index" to control the order of widgets in group then use values 1, 2, 3, ... as value 0 will always put the widget at the end.

elmonic, 2024-11-13:
OK, thank you, but if it starts from scratch, it is more basic.

elmonic, 2024-11-13:
OK, thank you, but if it starts from zero, it is more basic.

closed 2024-11-13 by mvladic