eez-open/studio #632: white screen

JJZBQA, 2024-11-13:
[image]
I copied the text Font from one project to another, and after running for a while, a white screen appeared. Regardless of whether I close it or open other projects, the same issue occurs — it runs for a while and then the screen turns white.

mvladic, 2024-11-13:
If this happens again open Developer Tools:

[image]

and then save the console log with the right click context menu:

[image]

JJZBQA, 2024-11-14:
![image](https://github.com/user-attachments/assets/0a5c9816-1c9c-4832-857b-ee1b4b0b4eb6
[image]
[-1731582849906.log](https://github.com/user-attachments/files/17747736/-1731582849906.log)
[-1731582956028.log](https://github.com/user-attachments/files/17747755/-1731582956028.log)
[-1731582966382.log](https://github.com/user-attachments/files/17747758/-1731582966382.log)

Happen again

mvladic, 2024-11-14:
What OS and Studio version you are using?

JJZBQA, 2024-11-14:
Windows11，Latest version
---- Replied Message ----
From Martin ***@***.***> Date 11/14/2024 19:23 To ***@***.***> Cc ***@***.***>***@***.***> Subject Re: [eez-open/studio] white screen (Issue #632) 
What OS and Studio version you are using?

mvladic, 2024-11-14:
> Windows11，Latest version

You compiled the latest version from the source?

mvladic, 2024-11-14:
> I downloaded the latest version you compiled.

Ok. I believe I fixed this problem. You can wait a several days until I release version 0.20.0 or you can build from source to test it now.

JJZBQA, 2024-11-14:
Thank you very much.And today, when I moved a component, I pressed the shortcut key to save it, and then my project was very big, which also triggered such a bug once.The preservation time has also become very long. ***@***.*** ***@***.*** ---- Replied Message ---- From Martin ***@***.***> Date 11/14/2024 21:15 To ***@***.***> Cc ***@***.***>***@***.***> Subject Re: [eez-open/studio] [LVGL] White screen while coping font from one project to another (Issue #632) I found a memory leak while coping font from one project to another. So, this could be the issue you are experiencing. It is fixed now.

closed 2024-11-18 by mvladic