eez-open/studio #634: [LVGL] bar with image background not showing on the esp32 but works in simulator

petrubuciu, 2024-11-14:
**Describe the bug**
i dont know if is a bug or not. i have an image (png) and i put over the image 2 bars with image background to animate the image, work perfect in the simulator but i upload the firmware to my esp32 doesnt show anything only the first image. i tried to to move those bars in the widget structure on every possible position as a parent and children.

this is from eez studio:
[image]
this are the widgets structure:
[image]
and this is from my esp:
[image]

**Desktop:**
 - OS: MacOs Sonoma
 - Version 19.0

** LVGL version**
- 9.2.2

if you need anything else please let me know and i will provide.

mvladic, 2024-11-14:
Did you forget to call `ui_tick()`? It must be called periodically - the best place is after `lv_timer_handler` or `lv_task_handler` call.

petrubuciu, 2024-11-14:
i call the  lv_timer_handler(); and ui_tick(); if i dont call the ui_tick() the ui doesnt work

mvladic, 2024-11-14:
Can you send me your `eez-project` file?

petrubuciu, 2024-11-14:
how can i send you? i dont want to put it here. thanks

petrubuciu, 2024-11-14:
sent, thx

mvladic, 2024-11-14:
I compiled your project using "Emscripten + SDL" with LVGL 9.2.2 and it works there too. I only changed bitmaps format to RGB888:

[image]

Here is also a new project that has only Bar widgets like in your project. This also works in both simulator and in compiled code.

[lvgl_test_anim.zip](https://github.com/user-attachments/files/17746878/lvgl_test_anim.zip)

[image]

Unfortunately, I don't have any ESP32 board where I can test your project with the LVGL 9.2.2. What board do you use? Is it custom board or some development board that can be bought?

mvladic, 2024-11-14:
Ups, sorry, I posted images from your project without asking you. I will delete it.

petrubuciu, 2024-11-14:
i will test now with rgb888 and report thx :)

mvladic, 2024-11-15:
I posted some findings to you as DM on Discord, but I think it also makes sense to post it here so anybody can see it. So, here it is:

I have this board https://github.com/lvgl/lv_port_renesas_rx72n-envision-kit which also has 16bit (RGB565) display, so I wanted to try this Bar animation test project. This is Renesas board that uses some kind of 2D graphics acceleration called Dave2D. With that acceleration enabled there are all kind of problems and I never managed, no matter what I tried, to run without problems this project (problems like not getting the expected result or even crashing). But, when I finally disabled Dave2D in `lv_conf.h` it suddenly started to work without any problem. I set all the images to RGB565A8 format. A8 is required because images must have an alpha channel because without it recoloring will not work. Here is how it looks:

https://github.com/user-attachments/assets/a86b81f8-a59e-4b1e-aa72-311a69207ac1

Even this Renesas board doesn't have anything in common with your ESP32 based board, except it also has 16 bit display, I wanted to share these findings with you because obviously not all drawing libraries works in LVGL even though they are part of LVGL. Dave2D can be found here in LVGL repository:

[image]

I suppose, without Dave2D, software rendering is used which can be found here:

[image]

Here is my eez-project that works on Renesas board:

[lvgl_test_bar_animations.zip](https://github.com/user-attachments/files/17777241/lvgl_test_bar_animations.zip)

closed 2025-02-04 by mvladic