eez-open/studio #644: background color not changed after save

shrikantvaishnav, 2024-11-21:
**Describe the bug**
After saving and buliding eez project for LVGL i found that from the next time onwards it doesnt allow to save background color.

**To Reproduce**
Steps to reproduce the behavior:
1.Create a project for LVGL
2. Click on Background than color
3. first time it will save
4. next time background will not save(may after app close and than open)

**Expected behavior**
It should also to edit everytime

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS:  Windows
 - Version 11 Pro

** LVGL version **
- 8.x 

** EEZ Studio version **
0.19.0

**Additional context**
No

mvladic, 2024-11-21:
I can't reproduce this bug. Here is how I tested:

[image]

shrikantvaishnav, 2024-11-22:
Sir you are using **EEZ with LVGL** while i am facing this issue for **LVGL Only project** also your EEZ version is .20.0 while in my case its .19.0.

mvladic, 2024-11-22:
I still can't reproduce it:

[image]

You are obviously doing something differently. Can you record a screen demonstrating this issue? I'm using [ScreenToGif](https://www.screentogif.com/) to record a screen.