eez-open/studio #648: Support for custom components.

devcarbon-com, 2024-11-25:
**Is your feature request related to a problem? Please describe.**
Currently, there's no functionality to create reusable custom components that can be updated globally. When you have multiple instances of the same design element across your project, any design changes need to be manually applied to each instance individually, which is time-consuming and prone to inconsistencies.

**Describe the solution you'd like**
Implement a system for creating and managing custom components similar to those found in Square Line Studio or modern web design tools. These components should:
- Be created once and reused throughout the project
- Update automatically across all instances when the master component is modified
- Maintain consistency throughout the design
- Allow for easy management of design systems and repeated elements

**Describe alternatives you've considered**
I cannot think of any good alternatives. 
  
**Additional context**
This feature could be particularly valuable for:
- Large-scale projects with recurring design elements
- Projects that need frequent design updates

zalexzperez, 2024-11-25:
Isn't this what User Widgets are there for?

devcarbon-com, 2024-11-25:
@zalexzperez  Ah, fantastic! Thank you, I missed that when I was looking through the docs.

closed 2024-11-25 by devcarbon-com