eez-open/studio #649: [LVGL] Add "screen stack control" property to Change Screen action

zalexzperez, 2024-11-25:
If we need to change screen without adding the active one to the stack, we need the missing Set Active Screen action. 
In order not to create a new action just for this, it makes sense to have a "Put active screen on the stack" option - enabled by default - on the Change Current screen action.

mvladic, 2024-11-27:
Implemented.

I added a new property called "Use stack":

[image]

zalexzperez, 2024-11-27:
The property seems to come disabled by default. And also, it's marked in red as if it requires user intereaction
[image]

mvladic, 2024-11-27:
> The property seems to come disabled by default.

Fixed.

closed 2024-11-30 by zalexzperez