eez-open/studio #650: [Question]The generated source code cannot find the variables and arrays created in EEZ

gao-zhida, 2024-11-26:
I have a question, why can't the variables created in EEZ be found in the compiled source code? Because I need to read their values on the microcontroller, but I can't find them in the source code, where should I find them？
[image]
[image]

mvladic, 2024-11-26:
Check this repository: https://github.com/eez-open/native-interface-lvgl-with-flow

Start from [Accessing flow variables from the native code](https://github.com/eez-open/native-interface-lvgl-with-flow?tab=readme-ov-file#accessing-flow-variables-from-the-native-code).

gao-zhida, 2024-11-26:
> 检查此存储库：[https ：//github.com/eez-open/native-interface-lvgl-with-flow](https://github.com/eez-open/native-interface-lvgl-with-flow)
> 
> [从从本机代码访问流变量](https://github.com/eez-open/native-interface-lvgl-with-flow?tab=readme-ov-file#accessing-flow-variables-from-the-native-code)开始。

The file I generated using EEZ does not have vars.cpp
[image]

mvladic, 2024-11-26:
File `vars.cpp` is not generated by the Studio, it is coded by hand for the purpose of demonstrating how to access flow variables from the code.

gao-zhida, 2024-11-26:
> File `vars.cpp` is not generated by the Studio, it is coded by hand for the purpose of demonstrating how to access flow variables from the code.

Understood, thank you！

guochenwei-sp, 2024-11-26:
Variables of NATIVE type can be used in code, but you need to implement the read and write functions yourself.

gao-zhida, 2024-11-26:
> NATIVE类型的变量可以在代码中使用，但是需要自己实现读写函数。

数组也是使用同样方法读写吗？

closed 2025-03-25 by mvladic