eez-open/studio #656: Style exports with wrong color format

lsroka76, 2024-12-04:
When I create Style and define some colors for it they are exported in wrong format in styles.c.

mvladic, 2024-12-04:
Can you give me some example, i.e. what did you expect and what did you get?

lsroka76, 2024-12-05:
I use "Add Style", name it like "test_style", select for "Button" and then select colors for background and gradient using Color Picker.
When I check hex values for colors in EEZ Studio everything is OK but after export project to C files in styles.c there are "lv_color_hex(0xffxxxxxx)" - it looks like color is exported with alpha mask?

I use LVGL 8.3.

mvladic, 2024-12-05:
> ... it looks like color is exported with alpha mask?

Yes, but it is ignored by the `lv_color_hex` function. Maybe we should omit alpha mask so instead of 0xffrrggbb just export 0xrrggbb to prevent any confusion.

> Also RGB order looks misplaced?

It is not misplaced. Function `lv_color_hex` expects argument in the format `0x??rrggbb` (where `??` is ignored) which is exactly what is generated.

lsroka76, 2024-12-08:
My fault - data lines for R and B were switched.

closed 2025-02-04 by mvladic