eez-open/studio #658: Allow quickly checking native action implementation if it has one

zalexzperez, 2024-12-06:
Just an idea to integrate native actions a little more.
Right click on the action like this:
[image]

And it should automatically take you to the highlighted action if it's found:
[image]

mvladic, 2024-12-06:
Where should Studio look for the implementation? Studio by default doesn't create `actions.cpp` or `actions.c`. So, for this feature to work we need some setting where user will specify in which file native actions are implemented.

zalexzperez, 2024-12-06:
You're right, I didn't think about that.
There should be a setting and if the user failed to provide a path, Studio will warn about it when using the Check implementation feature.