eez-open/studio #672: GUI Actions ( showpage ) not appearing in listt with a lvgl-flow projec ( appears when it is  a dashboard)

ghisbo, 2024-12-19:
**Describe the bug**
when i create a project with LVGL-FLOW, the  GUI Actions do not contain "SHOWPAGE"

**To Reproduce**
Steps to reproduce the behavior:
Create a project with LVGL-FLOW
Add a button
add an event ( type flow ) 
in the action list, the ShowPage is not available

**Expected behavior**
the showpage and some more should be there

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - Win
 - Version 11

** LVGL version 9.x

mvladic, 2024-12-19:
In the LVGL projects you should use LVGL action with the "Change Screen" for this:

[image]

[image]

[image]

ghisbo, 2024-12-19:
Perfect

closed 2024-12-19 by ghisbo