eez-open/studio #674: Loading Font/Image data from an external Flash memory (access in the RAM)

MyVanitar, 2024-12-20:
Hello,

I want to store the font/image data inside an external flash memory (binary or whatever method), not the memory-mapped method, so it's not a part of the internal memory to deal similar to an internal memory. So a function should read the data and load it all into the SRAM, then LVGL can access it.

How can I follow this in the EEZ Studio, I looked at the generated font file, it I stuck to do what :-)

mvladic, 2024-12-20:
Can you share how would you do that from C on your target platform? For example, if you want to change the font for the object then you need to call `lv_obj_set_style_text_font` and give a font pointer (`lv_font_t *`) to this function - what will you give as a font pointer in your case?

In Studio, you can add the font and use it to set widget text font. Then Studio will generate the following code (let assume that font name is `my_font`):

- The binary data of the font in the source file `ui_font_my_font.c`. When compiled this font data will be stored inside FLASH memory and LVGL will use this data from there (i.e. no SRAM will be used).
- `fonts.h` header file with the following declaration:
  ```extern const lv_font_t ui_font_my_font;```
- In the `screens.c`  (which includes `fonts.h`) it will generate the following line:
  ```lv_obj_set_style_text_font(obj, &ui_font_my_font, LV_PART_MAIN | LV_STATE_DEFAULT);```

Currently, this is the only way to use custom fonts in studio.

mvladic, 2024-12-20:
> When compiled this font data will be stored inside FLASH memory and LVGL will use this data from there

This can be changed if you define `LV_ATTRIBUTE_LARGE_CONST` in `lv_conf.h` and redirect from FLASH to some other memory, for example external flash.

MyVanitar, 2024-12-20:
I think I found a good option here: https://docs.lvgl.io/9.0/overview/font.html

So we can load all font data from an external Flash (*.bin file) into the SRAM (buffer) and then assign a font object to it. 

if you like you can add a feature in the EEZ for the user to select and the code generated by this selection.

The same thing should exist for images, I'm searching now.

guochenwei-sp, 2024-12-25:
This is also the feature I currently need. I modified some of the code in the project and generated bin and. h files in the following format from the image files.
[image]
[image]

MyVanitar, 2024-12-25:
> This is also the feature I currently need. I modified some of the code in the project and generated bin and. h files in the following format from the image files. 

I think what you did is for an external QSPI memory and memory mapped. However since QSPI, XIP, and memory mapping is not available for many MCUs, a lower level solution is to read an SPI external flash and fill another external SRAM with that data (addresses are known). then assign an LVGL object to the SRAM buffer, either an image or a font. 

if he could support both scenarios, it would be a very good feature.